PaulStoffregen/cores #46: Non Teensy board recognized as Teensy USB Serial with USB-Hub

NicoHood, 2015-01-11:
I noticed a very weird problem.

When I try to connect my HoodLoader via USB to an USB Hub it is recognized as Teensy Serial and the uploading via Arduino IDE doesnt work anymore (the veryfication fails but the sketch is uploaded correct). On a direct USB Port its recognized correct as 16u2.

Sometimes the CDC reset occurs and then the board isnt recognized at all anymore. If you replug it its still dead. Switching to a different usb hub port works well and then the 1st port works again.

What I tried:
I am using 1.6 rc1. I redownloaded a clean installation, maybe the teensy install on my other Arduino IDE messed something up, but the problem is still there.
I tried different Arduino boards and a HL2 firmware that worked on usb hubs.
I tried different PIDs
This problem only occured after i installed the teensyduino with its drivers. And i have no clue how to uninstall them.
I havent tried a different usb hub, since I dont have another one here.

PaulStoffregen, 2015-01-11:
Teensy using a "Compatible-Id" match, which is at a lower priority than "Hardware-Id" matching, and matches to all CDC protocol serial.  Details here:

https://forum.pjrc.com/threads/26265-Teensy-USB-driver-quot-hijacks-quot-Samsung-Galaxy-S4-phone-USB-function

If you have some non-Teensy device and you want it to appear with the correct name, simply install its own "driver".  Hardware-Id matching takes precedence over Compatible-Id.

However, on modern Windows, your driver MUST be signed.  Unsigned drivers are always treated as the least trusted, starting with Windows 7.

closed 2015-01-11 by PaulStoffregen

NicoHood, 2015-01-11:
I booted another Windows with no Teensy installed on the same PC.
The problems are the same, just that its not a teensy CDC device. Seems that my USB Hub is dying. I will try another soon.

I am using unsigned drivers, you can install them with that special reboot methode under window 8. Does this matter? Did you pay for the signed drivers or how can i create signed drivers?

PaulStoffregen, 2015-01-11:
https://forum.pjrc.com/threads/26265-Teensy-USB-driver-quot-hijacks-quot-Samsung-Galaxy-S4-phone-USB-function?p=61275&viewfull=1#post61275

NicoHood, 2015-01-11:
Thx for that very useful information. That ceritficate is too expensive for me to provide free software. But now Its clear to me, I never knew how to sign drivers :)

I tested the uploading under Ubuntu and I had the same problems. My usb port is nearly dead for sure.

NicoHood, 2015-01-11:
Another usb hub works fine (usb 3). But it still shows up as teensy on the usb hub but with direct connection it shows up correctly. Any idea why

another usb2 hub doesnt work again. the first i mentioned was also usb2

I am wondering if this could be a problem of my bootloader and the watchdog reset? Or my PC itself has a problem? the 2nd hub shows exactly the same behaviour. All ports on my pc are usb3.

PaulStoffregen, 2015-01-14:
Any interest in developing USB features on Teensy 3.1?  Seems like you've been putting a lot of work into Leonardo, only resulting in frustration.

Teensy 3.1 has a LOT more memory and capability, so we're not limited in the kinds of code and features that can be accepted.

NicoHood, 2015-01-14:
Someone donated me a Teensy 3.1 board this month. I will be back home on feb 2, then i can test it. Ive looked through the source here and there and it looks pretty good and more flexible. I really like it. I just dont like the way how to install it. A simple hardware folder would be better for me, because it really changes the ide a lot in the menu. And i never used a 32bit arm. But we will see.

The reason why i use leonardo/16u2 is just that I wanted to get more out of a simple Arduino. I think its great what great features you can add to a simple Uno. Nobody used the 16u2 before. HoodLoader2 was just for fun you could say and a challenge. If I know more about Teensy I'd also like to port the teensy usb core to the old 16u2 as well.

If you want to contact me, the best way to do so is here:
https://nicohood.wordpress.com/about/

PaulStoffregen, 2015-01-14:
On 32 bit ARM with DMA-based USB, so much more should be possible.  Later this year, I'm going to try USB networking,  Someone on the forum has already put some good work into USB audio, which I'd really like to merge when it's stable (I'll get personally involved in that code in a few months).

There's enough RAM and Flash, and we have the USB Type menu, so accepting code advanced USB features into the code is a lot easier than Arduino, where there's tons of cooks in the kitchen and many of them are more concerned with saving dozens of bytes of memory than making awesome features many people can use.

If you want to try this stuff on Teensy 3.1, let me know.  If you want to fiddle with USB audio, I could send you one of the audio adaptor boards.

NicoHood, 2015-01-15:
Sounds great. I have to work on that when I am back home. I am very busy now (this is not my job, its a hobby I started one year ago) so I cannot tell when and if I will work on that. I also never did USB audio or anything else far away from cdc and hid. You should know how complicated usb is and what errors it can cause ;) I could imagine something like a custom midi board with audio. Like these professional midi board friend of mine has got.

If you wanna send me one of the adaptors feel free to contact me, but I cannot guaranty that I have time or that I am able to produce something useful.

I just have two question to you Paul:
am I allowed to provide the Teensy drivers with my software (because of the great signed, generalized driver) or am I only allowed to link it?

There is still some documentation on the website about Teensy 1 and 1++. Is there any reason why Its dead? I'd like to know if its still possible to compile sketches with the teensy core for a at90usb162. In the github repo there is also no boards.txt but its mentioned in the readme. Is the github source up to date?