PaulStoffregen/cores #53: Error using sscanf function

rummanwaqar, 2015-02-16:
Using sccanf generates the following error:
```
/Applications/Arduino-1.0.6.app/Contents/Resources/Java/hardware/tools/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/4.7.2/../../../../arm-none-eabi/lib/thumb2/libc.a(lib_a-writer.o): In function `_write_r':
writer.c:(.text+0x16): undefined reference to `_write'
collect2: error: ld returned 1 exit status
```

PaulStoffregen, 2015-02-16:
Please post a small but complete program that reproduces this error.

rummanwaqar, 2015-02-16:
```
#include <ArduinoHardware.h>

#define MAX_LEN 200

char line[MAX_LEN];
const char* format = "$C%fP%fR%fT%fMx%fMy%fMz%fAx%fAy%fAz%f";

void setup() {
  Serial1.begin(115200);
}

void loop() {
  char checksum;
  if(Serial1.available()) {
    int rv = Serial1.readBytesUntil('\r\n', line, MAX_LEN);
    int len = strnlen(line, MAX_LEN);
    rv = sscanf(line+ (len-4), "*%hhx", &checksum);
  }
}
```

Error:
```
/Applications/Arduino-1.0.6.app/Contents/Resources/Java/hardware/tools/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/4.7.2/../../../../arm-none-eabi/lib/thumb2/libc.a(lib_a-writer.o): In function `_write_r':
writer.c:(.text+0x16): undefined reference to `_write'
collect2: error: ld returned 1 exit status
```

rummanwaqar, 2015-02-17:
@PaulStoffregen were you able to duplicate the error, and know why it is being generated? The most peculiar thing is that `sscanf` shouldn't call `_write`.

PaulStoffregen, 2015-02-17:
Please post a complete program.

    sketch_feb17a.ino:1:29: fatal error: ArduinoHardware.h: No such file or directory
    compilation terminated.
    Error compiling.

PaulStoffregen, 2015-02-17:
Looks like this is also on the forum.  (so far, neither links to the other)

https://forum.pjrc.com/threads/27573-Teensy-3-1-and-sscanf

PaulStoffregen, 2015-02-17:
Please test with 1.21-beta5.  Looks like this problem may already be fixed.

https://forum.pjrc.com/threads/27740-Arduino-1-6-0-any-plans-to-support-it?p=64839&viewfull=1#post64839

rummanwaqar, 2015-02-17:
Thanks, the error is fixed in this beta.

closed 2015-02-17 by rummanwaqar