PaulStoffregen/cores #54: SERIAL_PORT_MONITOR incorrect when no USB

xxxajk, 2015-02-18:
teensy/cores/teensy3/pins_arduino.h...

```
#define SERIAL_PORT_MONITOR             Serial
```
This is all fine, but, should default to Serial1 when USB is disabled, and the other defines should change as well...

PaulStoffregen, 2015-02-18:
That's a very good point.

I'm packaging up beta6 right now, so this will have to wait for a bit.

xxxajk, 2015-02-18:
I can wait for beta 7 or later. I just thought I would bring this to your
attention since it would a fact mod codeI can wait for beta 7 or later. I
just thought I would bring this to your attention since it would affect
host mode code.

xxxajk, 2015-02-18:
Looks like Gmail and voice typing has problems but I think you understand.

PaulStoffregen, 2015-06-18:
fixed
https://github.com/PaulStoffregen/cores/commit/37e24b01079145aed9d53609c7140a5989254c59

closed 2015-06-18 by PaulStoffregen