PaulStoffregen/cores #67: LICENSE file

AlexandreZani, 2015-04-04:
The licensing for this project is hard to read. Different files have different license headers and many files don't have a licensing header at all. That makes it hard to understand how to treat this code. (Also, some people have IP assignment agreements with their employers and need a clearer license before we can offer contributions) Could you put a LICENSE file at the top of the tree?

PaulStoffregen, 2015-04-09:
A top level file will only get out-of-date.  While this might be nice, I think over the long term having inaccurate, not-updated info is worse, so I'm not very inclined to do this.

closed 2015-04-09 by PaulStoffregen

AlexandreZani, 2015-04-09:
It also makes it impossible to know what the license is for files which do
not have a license header though. This is an obstacle to people using the
code you provide.

PaulStoffregen, 2015-04-09:
which specific files are missing license headers?

AlexandreZani, 2015-04-09:
teensy3/avr/* and teensy3/util/* for instance