PaulStoffregen/cores #79: DMAChannel destinationCircular bug

pedvide, 2015-06-03:
I think there's a bug in the destinationCircular/sourceCircular functions for the Teensy 3.x.
The DMOD (or SMOD) is set to (31 - __builtin_clz(len)), but it should be (32 - __builtin_clz(len)). Otherwise it only fills half of the buffer before going back to the 1st element.
I've only tested it with destinationCircular, it works correctly if I call the function with 2*len instead.

duff2013, 2015-06-04:
I've been using destinationCircular/sourceCircular in my UartEvent library for awhile now and it works fine with uint8_t data. I've checked address's of the SADDR and DADDR registers all the way through the buffer and they allways wrapped correctly for me. Maybe you can post how you setup your DMA? I found that you need your buffer aligned to proper byte boundry for it to work correctly.

pedvide, 2015-06-04:
Sorry, I'll post the code when I get home in a few hours.
Thinking about this I've also realized that maybe I don't need a circular buffer implementation, maybe a regular buffer will be enough.
My goal is to write a class to handle a circular buffer that gets filled by DMA when the ADC finishes a conversion. The idea is that the user will only see functions like read(), isFull(), isEmpty().
I use a circular implementation in DMA so I can update the buffer's pointers every time the DMA finishes a transfer (so I have transfer counts of 1 and DMA interrupt every transfer).
To get a proper aligned array on the heap I reserve a bit more space with malloc and move the pointer until it's aligned. So maybe what I'm seeing is indeed a problem with the alignment of the array.

pedvide, 2015-06-04:
The code is:
```
dmaChannel->source(ADC0_RA);
//      TCD->SADDR = ADC_RA;
//		TCD->SOFF = 0;
//		TCD->ATTR_SRC = 2; // upper 8 bits of TCD->ATTR are TCD->ATTR_SRC
//		TCD->NBYTES = 4;
//		TCD->SLAST = 0;

    dmaChannel->destinationCircular(p_elems, 2*b_size);
//    TCD->DADDR = p_elems;
//    TCD->DOFF = 2;
//    TCD->ATTR_DST = ((31 - __builtin_clz(len)) << 3) | 1;
//    TCD->NBYTES = 2;
//    TCD->DLASTSGA = 0;
//    TCD->BITER = len / 2; // we'll overwrite them
//    TCD->CITER = len / 2;

    dmaChannel->transferSize(2);
//      TCD->NBYTES = 2;
//	    TCD->ATTR = (TCD->ATTR & 0xF8F8) | 0x0101;

    dmaChannel->transferCount(1);
//    TCD->BITER = 1;
//    TCD->CITER = 1;

    dmaChannel->interruptAtCompletion();

dmaChannel->triggerAtHardwareEvent(DMAMUX_SOURCE_ADC0); // start DMA channel when ADC finishes a conversion
	dmaChannel->enable();
	dmaChannel->attachInterrupt(call_dma_isr);

```

call_dma_isr just calls void_isr().

```
void_isr() {
    //digitalWriteFast(LED_BUILTIN, !digitalReadFast(LED_BUILTIN));
    write(); // move the circular buffer's pointers
    dmaChannel->clearInterrupt();
}
```

The p_elems buffer is:
```
Buffer: Address, Value
1FFFEA60, 0
1FFFEA62, 0
1FFFEA64, 0
1FFFEA66, 0
1FFFEA68, 0
1FFFEA6A, 0
1FFFEA6C, 0
1FFFEA6E, 0
```

If I convert values the buffer is filled from the top, to the end and continues from the top again. But only if I use `dmaChannel->destinationCircular(p_elems, 2*b_size);` , not `dmaChannel->destinationCircular(p_elems, b_size);`

pedvide, 2015-06-04:
Sorry, I forgot to add how I align the memory:
```
    // len must be power of two. extract the alignment
    alignment = 32 - __builtin_clz(len);

    // calculate mask
    uint8_t mask = (1 << (alignment)) - 1;

    // To align to 16 bytes, allocate 15 bytes more and "move" the pointer to a 16 byte alignment, same for other sizes
    // we need to free(p_mem), the original pointer returned by malloc.
    // see http://stackoverflow.com/questions/227897/solve-the-memory-alignment-in-c-interview-question-that-stumped-me/
    p_mem = malloc(len+alignment);
    p_elems = (int16_t*)(((uintptr_t)p_mem+len) & ~ (uintptr_t)mask);
```
I have to do this because there's no memalign (or similar) function in the libraries included.

duff2013, 2015-06-04:
I aligned my buffer using a builtin attribute such as -> __attribute__((aligned("Your Buffer Size in Bytes, Power of 2")))myBuffer[]. 

The destinationCircular function and transferSize function both modify the ATTR register, maybe this is what is halving your buffer, I let destinationCircular function set my attributes and I do not call transferSize function. Here you can see how i set the [buffer], and the DMA setup [code].

[buffer]:https://github.com/duff2013/UartEvent/blob/master/utility/Uart1Event.cpp#L47
[code]:https://github.com/duff2013/UartEvent/blob/master/utility/Uart1Event.cpp#L153

pedvide, 2015-06-05:
I'm going to close this as I think I have alignment issues (I mean my arrays :P ).

pedvide, 2015-06-05:
Well I thought I could close it myself, Paul, can you close this? It's not a bug.

PaulStoffregen, 2015-06-05:
Ok, closing this one.

closed 2015-06-05 by PaulStoffregen