PaulStoffregen/cores #111: Compilation error using gcc-arm-embedded-4.9.3 in teensy3 using provided makefile without arduino

rzetterberg, 2016-01-23:
I'm trying to use my teensy 3.1 without Arduino on Linux. But I'm getting a compile error using gcc-arm-embedded-4.9.3. I'm running `make` inside the `cores/teensy3` directory without arduino. Here's the compiler error output:

```
arm-none-eabi-g++ -std=gnu++0x -felide-constructors -fno-exceptions -fno-rtti -Wall -g -Os -mcpu=cortex-m4 -mthumb -MMD -DF_CPU=48000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -DUSING_MAKEFILE -D__MK20DX256__ -DARDUINO=10600 -DTEENSYDUINO=121 -I.  -c -o Stream.o Stream.cpp
In file included from WProgram.h:14:0,
                 from Arduino.h:1,
                 from Stream.cpp:23:
avr_functions.h:97:104: error: 'char* utoa(unsigned int, char*, int)' was declared 'extern' and later 'static' [-fpermissive]
 static inline char * utoa(unsigned int val, char *buf, int radix) __attribute__((always_inline, unused));
                                                                                                        ^
In file included from /nix/store/a10h1v4iyw1m4kcxzqb0dgjzzkw42fm2-gcc-arm-embedded-4.9-2015q1-20150306/arm-none-eabi/include/stdlib.h:11:0,
                 from WProgram.h:4,
                 from Arduino.h:1,
                 from Stream.cpp:23:
/nix/store/a10h1v4iyw1m4kcxzqb0dgjzzkw42fm2-gcc-arm-embedded-4.9-2015q1-20150306/arm-none-eabi/include/stdlib.h:184:8: note: previous declaration of 'char* utoa(unsigned int, char*, int)'
 char * _EXFUN(utoa,(unsigned, char *, int));
        ^
In file included from WProgram.h:14:0,
                 from Arduino.h:1,
                 from Stream.cpp:23:
avr_functions.h:99:95: error: 'char* itoa(int, char*, int)' was declared 'extern' and later 'static' [-fpermissive]
 static inline char * itoa(int val, char *buf, int radix) __attribute__((always_inline, unused));
                                                                                               ^
In file included from /nix/store/a10h1v4iyw1m4kcxzqb0dgjzzkw42fm2-gcc-arm-embedded-4.9-2015q1-20150306/arm-none-eabi/include/stdlib.h:11:0,
                 from WProgram.h:4,
                 from Arduino.h:1,
                 from Stream.cpp:23:
/nix/store/a10h1v4iyw1m4kcxzqb0dgjzzkw42fm2-gcc-arm-embedded-4.9-2015q1-20150306/arm-none-eabi/include/stdlib.h:183:8: note: previous declaration of 'char* itoa(int, char*, int)'
 char * _EXFUN(itoa,(int, char *, int));
        ^
<builtin>: recipe for target 'Stream.o' failed
make: *** [Stream.o] Error 1
```

If I add `-fpermissive` to `CPPFLAGS` I get a linker error instead:

```
arm-none-eabi-gcc -Os -Wl,--gc-sections,--defsym=__rtc_localtime=0 --specs=nano.specs -mcpu=cortex-m4 -mthumb -Tmk20dx256.ld -o main.elf usb_m
erial3.o usb_dev.o usb_keyboard.o keylayouts.o usb_midi.o usb_joystick.o ser_print.o pins_teensy.o usb_rawhid.o nonstd.o eeprom.o usb_mem.o us
usb_desc.o serial1.o mk20dx128.o serial2.o analog.o math_helper.o touch.o usb_seremu.o usb_serial.o new.o HardwareSerial1.o DMAChannel.o Print
nst.o Stream.o IPAddress.o Tone.o AudioStream.o yield.o main.o usb_flightsim.o HardwareSerial3.o WString.o HardwareSerial2.o avr_emulation.o W
ntervalTimer.o -lm
Print.o: In function `utoa':
/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:98: multiple definition of `utoa'
HardwareSerial1.o:/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:98: first defined here
Print.o: In function `itoa':
/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:100: multiple definition of `itoa'
HardwareSerial1.o:/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:100: first defined here
usb_inst.o: In function `utoa':
/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:98: multiple definition of `utoa'
HardwareSerial1.o:/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:98: first defined here
usb_inst.o: In function `itoa':
/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:100: multiple definition of `itoa'
HardwareSerial1.o:/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:100: first defined here
Stream.o: In function `utoa':
/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:98: multiple definition of `utoa'
HardwareSerial1.o:/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:98: first defined here
Stream.o: In function `itoa':
/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:100: multiple definition of `itoa'
HardwareSerial1.o:/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:100: first defined here
IPAddress.o: In function `utoa':
/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:98: multiple definition of `utoa'
HardwareSerial1.o:/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:98: first defined here
IPAddress.o: In function `itoa':
/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:100: multiple definition of `itoa'
HardwareSerial1.o:/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:100: first defined here
Tone.o: In function `utoa':
/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:98: multiple definition of `utoa'
HardwareSerial1.o:/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:98: first defined here
Tone.o: In function `itoa':
[... 19 lines cut ...]
HardwareSerial1.o:/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:100: first defined here
HardwareSerial3.o: In function `utoa':
/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/HardwareSerial.h:174: multiple definition of `utoa'
HardwareSerial1.o:/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:98: first defined here
HardwareSerial3.o: In function `itoa':
/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/HardwareSerial.h:175: multiple definition of `itoa'
HardwareSerial1.o:/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:100: first defined here
WString.o: In function `utoa':
/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:98: multiple definition of `utoa'
HardwareSerial1.o:/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:98: first defined here
WString.o: In function `itoa':
/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:100: multiple definition of `itoa'
HardwareSerial1.o:/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:100: first defined here
HardwareSerial2.o: In function `utoa':
/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/HardwareSerial.h:174: multiple definition of `utoa'
HardwareSerial1.o:/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:98: first defined here
HardwareSerial2.o: In function `itoa':
/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/HardwareSerial.h:175: multiple definition of `itoa'
HardwareSerial1.o:/home/zed/development/embedded_c/teensy3_test/tmp/teensy-template/cores/teensy3/avr_functions.h:100: first defined here
collect2: error: ld returned 1 exit status
Makefile:92: recipe for target 'main.elf' failed
make: *** [main.elf] Error 1
```

I can supply more information if needed!

rzetterberg, 2016-01-23:
I can confirm that I can successfully compile the project using version 4.7.4:

```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.7.4 20130913 (release) [ARM/embedded-4_7-branch revision 202601]
```

I guess something significantly changed in 4.9.x.

FrankBoesing, 2016-01-23:
Yes, the toolchain was updated the mentioned functions were added. 
Pehraps there should be a #if GCC_VERSION <= xyz to prevent such errors.

wingunder, 2016-01-27:
@rzetterberg: Try the commit that I made in the following pull request: https://github.com/PaulStoffregen/cores/pull/104

@FrankBoesing: I think checking with the GCC_VERSION will be a hack, as it's really the version of newlib that matters in this case. Newlib introduced the `__NEWLIB__` and `__NEWLIB_MINOR__` macros in newlib version 2.2.0, so checking with these will be better in the future. However, using these checks for newlib versions < 2.2.0, will be very limited, i.e. using a #ifndef.

I would say we have 2 choices here:

1. Keep track of the newlib implementation and guard stuff with the `__NEWLIB__` and `__NEWLIB_MINOR__` macros.
2. Add weak attributes to existing functions that newlib decide to implement. This will come at the cost(?) of un-inlining funcions, if, and only if, they where previously inlined. Gcc requires that weak function are _not_ inlined.

Solution 2 (pull request https://github.com/PaulStoffregen/cores/pull/104) is obviously easier to maintain and less error prone, but I'm not sure how much the un-inlining, in this case, will hurt. BTW, has someone got any figures/info on this?

Another advantage of solution 2 is that if is a tool chain independent fix. If some tool chain uses something else than newlib, and cores wants to support it, solution 1 will be even more of a challenge.

PaulStoffregen, 2016-01-27:
I've merged #83. Does that adequately address this issue too?

wingunder, 2016-01-29:
@PaulStoffregen Yes, thanks, it works for both my tool chains, using std=gnu++0x.

closed 2016-01-30 by PaulStoffregen