PaulStoffregen/cores #117: Open HalfKay Bootloader source

NicoHood, 2016-02-26:
I'd like to kindly ask and discuss for opening the Teensy 2.0 USB HID HalfKay Bootloader source.
Since Teensy2.0 does not get much more updates and Teensy 3.x (and LC) is now the most up to date and fancy Teensy, you might consider to open the source for the halfkay bootloader for the avr product.

#### Pro
* People could recover their broken Teensy Bootloader (if they've overwritten it).
* The Bootloader can be used a reference to code a very small USB HID bootloader, similar to optiboot, but for HID.

#### Con
* Chinese people can copy it without caring on copyrights

Any thoughts from Paul and the Community about that?

PaulStoffregen, 2016-02-26:
Inspired by how iD software releases the source for old games like Doom & Quake, I'm considering doing the same when we eventually discontinue Teensy 2.0.

Until then, of course this doesn't work with PJRC's business model.  I know you and others would very much like to have this code as open source.  So would the many (not Chinese) distributors who resell Teensy.

NicoHood, 2016-02-26:
Yeah exactly. That was the idea. I just dont know when you consider Teensy 2 discontinued. I remember that you said you wont update the USB Core stuff again, and also I dont see anyone still use a Teensy2, cause a Teensy3 is still a Teensy, but better at mostly the same price. I dont know how good it still sells though. Is there anything more specific on your time line?

PaulStoffregen, 2016-02-26:
Teensy 2.0 is indeed selling.  We have substantial inventory and scheduled purchases well into 2017.

Eric-L-T, 2016-03-12:
> Until then, of course this doesn't work with PJRC's business model. I know you and others would very much like to have this code as open source. So would the many (not Chinese) distributors who resell Teensy.

@PaulStoffregen Can you please elaborate?

PaulStoffregen, 2016-03-12:
Sorry, I just don't have time to write a detailed reply.

This issue tracker is for reporting bugs in the Teensyduino code, and for suggesting & requesting features.  I really need to close this (and other) issues which aren't about specific bugs or features.

closed 2016-03-12 by PaulStoffregen