PaulStoffregen/cores #119: Wrong SET_ADDRESS implementation

NicoHood, 2016-03-30:
Referring to:
https://github.com/PaulStoffregen/cores/blob/d5a1b9745bc2935d0aa09294fe9e98b475b1e8cf/usb_serial/usb.c#L441-L446

This also applies to all other files with this implementation.
You first have to set the address and then enable it and not all at once. It is clearly stated that this mechanism (like in the teensy code) is wrong.

The 32u4 Datasheet says (section 22.7, page 269)
```
The USB device address is set up according to the USB protocol:
• the USB device, after power-up, responds at address 0
• the host sends a SETUP command (SET_ADDRESS(addr)),
• the firmware handles this request, and records that address in UADD, but keep ADDEN
cleared,
• the USB device firmware sends an IN command of 0 bytes (IN 0 Zero Length Packet),
• then, the firmware can enable the USB device address by setting ADDEN. The only accepted
address by the controller is the one stored in UADD.
ADDEN and UADD shall not be written at the same time.
UADD contains the default address 00h after a power-up or USB reset.
```

A possible implementation (modified from LUFA) could look like this:
```cpp
// 32u4 datasheet section 22.7
static void USB_Device_SetAddress(void)
{
  // Get address value
	uint8_t DeviceAddress = (USB_ControlRequest.wValue & 0x7F);

  // Record Address, keep ADDEN cleared
	//USB_Device_SetDeviceAddress(DeviceAddress);
  UDADDR = (UDADDR & (1 << ADDEN)) | (DeviceAddress & 0x7F);

  // TODO move upwards?
	Endpoint_ClearSETUP();

  // Send IN ZLP
	Endpoint_ClearStatusStage();

  // Wait for IN endpoint to get ready
	while (!(Endpoint_IsINReady()));

  // Enable USB device address
	//USB_Device_EnableDeviceAddress(DeviceAddress);
  UDADDR |= (1 << ADDEN);
}
```

Correct me if I am wrong

cc @limpkin (also applies to Mooltipass)

limpkin, 2016-06-25:
changed in Mooltipass source code.

NicoHood, 2016-06-26:
@limpkin Can you please reference the commit so @PaulStoffregen can also adapt the fix real quick. Thanks :)

PaulStoffregen, 2016-06-26:
Has anyone looked into whether this actually causes any problem?

limpkin, 2016-06-26:
Tested and working on my computer, will run the USB test suite on monday and will deploy to beta testers today.

PaulStoffregen, 2016-06-26:
What I mean is... despite the words "ADDEN and UADD shall not be written at the same time" on page 268, the fact is this code has been used by well over 100000 people over the last 8 years.  So far there's no indication it's ever caused any enumeration failures.

Or is there?  Is there any confirmed case where this code fails, but the modified version works?

limpkin, 2016-06-26:
I actually do agree with you and usually don't change code that has always been working... but I do tend to follow the manufacturer's guidelines.

PaulStoffregen, 2016-06-26:
Let's take a moment to ponder Dean's code above, which Nico offered as a possible implementation.  Specifically this line:

        UDADDR = (UDADDR & (1 << ADDEN)) | (DeviceAddress & 0x7F);

Now as a practical matter, no USB host controllers have drivers than send set address more than once without a reset event.  So there's no point to reading the ADDEN bit, since it'll always be zero.  So this is pointless.  But if it weren't pointless, it'd be even *worse*.  This would change the address while in the middle of a control transfer sequence (setup-in-ack).

Likewise, the advice recommends "records that address in UADD, but keep ADDEN cleared".  But ADDEN is later documented as a set-only bit.  Again, in practice hosts send only 1 set address.  But if we're considering what would be a good course of action, keeping the value in memory or a register until after the ACK phase is much more reasonable.  I'm pretty sure whoever wrote that part of the datasheet was describing the ancient & horrible example code Atmel originally published for these chips.

I will admit there doesn't seem to be any downside to performing 2 writes to UDADDR, first the low 7 bits with the MSB clear, and then the same thing again with MSB set.  Even if it's unnecessary, I'll probably make this change.

But I still want to know if **anyone** has found any case where this actually make a difference?

Again, I'd like to remind you this is extremely mature code that's been very widely deployed.  Never has there been any issue with the setting the address during enumeration.  So I'm a bit skeptical, and I certainly do not feel this is urgent.

limpkin, 2016-06-26:
Note that I never said I implemented it as Nicohood suggested, see from 14hours ago: https://github.com/limpkin/mooltipass/commit/e4500e384661940f8d8de8f8e744c1c4e671da51

And to answer your question: no, we never had enumeration problems on 2k+ devices.
I do share your opinion, which is why I only tackled it 3 months after Nicohood created this "issue".

PaulStoffregen, 2016-06-26:
Oh, I see you're writing UDADDR before sending the IN and getting the ACK.  I don't care what the datasheet says, that just seems like a bad approach to me.  I don't care if Dean's doing it in LUFA.  I'm absolutely not going to do that in Teensyduino's code.

I probably will do to 2 writes to UDADDR after the ACK, for version 1.30 in a few months.  But unless any case turns up where this actually matters, I'm putting it on my low priority list.  The code is extremely stable and well tested, so I really do not believe there's any real issue here.

limpkin, 2016-06-26:
Well, from the datasheet, ADDEN : "Set to activate the UADD (USB address)."
As you said, there'll never 2 consecutive set address commands between USB resets (and I agree) so it actually doesn't matter when you write UADD... or I missed something.

You don't need to emphasize the low priority aspect and "issue" definition when I concur...

PaulStoffregen, 2016-06-26:
While I've never seen any host do it, there is language in the USB spec that implies more than 1 set address is possible.  For example, on page 246 of the USB 2.0 spec:

    After successful completion of the Status stage, the device is allowed a SetAddress() recovery interval of
    2 ms. At the end of this interval, the device must be able to accept Setup packets addressed to the new
    address. Also, at the end of the recovery interval, the device must not respond to tokens sent to the old
    address (unless, of course, the old and new address is the same).

What happens in error conditions with the final ACK might also be a good question....