PaulStoffregen/cores #122: GCC 5.3 random()

FrankBoesing, 2016-04-11:
Just installed GCC 5.3 and got the following error:

C:\Arduino\hardware\teensy\avr\cores\teensy3/WProgram.h:51:21: error: ambiguating new declaration of 'uint32_t random()'

 uint32_t random(void);

                     ^

In file included from c:\arduino\hardware\tools\arm5\arm-none-eabi\include\stdlib.h:11:0,

                 from C:\Arduino\hardware\teensy\avr\cores\teensy3/WProgram.h:4,

                 from C:\Arduino\hardware\teensy\avr\cores\teensy3/Arduino.h:1,

                 from f:\build4def553c7c5cc6638db6fb6a30be3b17.tmp\sketch\coremark.h:27,

                 from f:\build4def553c7c5cc6638db6fb6a30be3b17.tmp\sketch\core_list_join.cpp:20:

c:\arduino\hardware\tools\arm5\arm-none-eabi\include\stdlib.h:231:6: note: old declaration 'long int random()'

 long _EXFUN(random,(_VOID));

      ^

FrankBoesing, 2016-04-11:
In addition, for -Og and -Os, fcvt() and fcvtf() are not extistant (used in nonstd.c by dtostrf()

tulth, 2016-06-14:
In teensy3/WProgram.h , I commented out this declaration:
 uint32_t random(void);
As a work around.  

WMath seems to conflict with arm-none-eabi stdlib.h

Ideas?  Rename random functions in WProgram.h?
Make the renaming of WMath random optional with it defaulted off?

closed 2016-10-27 by PaulStoffregen