PaulStoffregen/cores #131: License missing on crc16.h

matthijskooijman, 2016-05-29:
The file [util/crc16.h](https://github.com/PaulStoffregen/cores/tree/master/teensy3) in the teensy3 core is missing a license and copyright header. Looking at the repository-wide README, I didn't find any default license, it only refers to the comments in individual files.

I'd like to [include this code](https://github.com/matthijskooijman/arduino-dsmr/pull/3) into a library of mine, but cannot really do so without a proper license.

matthijskooijman, 2016-05-29:
`parity.h` and `delay.h` in the same directory are also missing a license header, though these are so simple that they're likely not quite copyrightable in the first place.

FrankBoesing, 2016-05-31:
I think they are from the AVR-LIBC http://www.nongnu.org/avr-libc/ 
Pls. correct me if I'm wrong

matthijskooijman, 2016-06-01:
@FrankBoesing, I don't think so. The API is copied from avr-libc, but the [implementation in avr-libc](http://svn.savannah.nongnu.org/viewvc/trunk/avr-libc/include/util/crc16.h?revision=2398&root=avr-libc&view=markup) is in AVR assembly. These files have been written to implement the same API in a cross-platform manner, for use on non-AVR platforms.

dhylands, 2016-06-01:
Well the comments of the avr-libc headers contain a C implementation, and that appears to be identical (not that there are very many ways of implementing a CRC-16).

matthijskooijman, 2016-06-01:
@dhylands, excellent point! Even without syntax highlighting, I completely ignored the comments when looking at that code :-)

Looking more closely, the teensy versions are indeed identical, except for changing `i` from `int` to `unsigned int` and expanding the `lo8` and `hi8` macros AFAICS.

FrankBoesing, 2016-06-02:
If you want a CRC with defined license - and which is way faster on teensy 3 - you could use this one: https://github.com/FrankBoesing/FastCRC

matthijskooijman, 2016-06-03:
@FrankBoesing, I mostly wanted one that is portable, and I rather want a small one than a fast one. I had already considered your library :-)

PaulStoffregen, 2016-06-03:
License info added.  Sorry for the delay, returned from a 5-day vacation and still many things to catch up.

closed 2016-06-03 by PaulStoffregen

matthijskooijman, 2016-06-07:
@PaulStoffregen, no problem, thanks for picking this up :-D

commit ce5692f9 (2016-06-03): Add license info for crc16.h
fixes #131