PaulStoffregen/cores #134: Distorted audio when using USB audio input

dturner, 2016-06-10:
When using the teensy as a USB audio device I get distorted audio from both the DAC and using the audio shield after around 2 minutes of playing a song. Once the distortion starts it gets progressively worse until no audio can be heard.

Steps to reproduce
- Set up a Teensy so it can ouput audio, either by connecting a speaker/buzzer to the DAC pin or using the Audio Shield and headphones
- In the Arduino IDE set the Teensy to USB type Audio
- Load the PassThroughUSB example
- From your computer choose 'Teensy Audio' as the audio interface
- Play a song on your computer. Personally I use "DJs take control" by SL2.
- Listen to the track

Expected results
- The track plays with no distortion

Actual results
- After a few minutes distortion occurs and gets progressively worse until only glitches are heard

Test environment/equipment
- Teensy 3.1
- Audio Shield Rev B with headphones
- Mac OSX 10.11.5
- Teensy Audio library: latest from github
- Teensy core library: latest from github

My suspicion is that this issue is caused by some kind of clock drift or synchronisation problem. Perhaps there's a bug in [usb_audio.cpp](https://github.com/PaulStoffregen/cores/blob/master/teensy3/usb_audio.cpp). I tried uncommenting various serial print statements in that file to debug further but have been unable to see any of those statements in the serial output - would welcome some advice on how to switch that on. 

I originally thought this was an issue with the audio shield so a duplicate bug can be found here: https://github.com/PaulStoffregen/Audio/issues/185

dturner, 2016-06-11:
Forum post on this issue: https://forum.pjrc.com/threads/34855-Distorted-audio-when-using-USB-input-on-Teensy-3-1?p=106490

dturner, 2016-06-12:
Workaround (which still results in some audio glitches) is to remove the USB SYNC_ENDPOINT: https://github.com/dturner/cores/commit/2ba638a3f791b26953f4835ff7e3b8d54ca129e8

PaulStoffregen, 2016-06-12:
Please stop double posting.  Let's use the forum for now.