PaulStoffregen/cores #197: Fast writing to SPI causes audio problems

FrankBoesing, 2016-11-14:
Hi Paul, 
please have a look here:

https://forum.pjrc.com/threads/36756-Teensyduino-1-30-Beta-4-Available?p=123627&viewfull=1#post123627

Perhaps, writing to other devices causes the same problems. Is there a way to give DMA priority for bus-accesses ?

FrankBoesing, 2016-11-14:
t... my fault.
!"§$%&/()=

Sorry.

FrankBoesing, 2016-11-14:
..actually, there IS a problem.. but hard to identify. Something with SPI.

FrankBoesing, 2016-11-14:
The problematic line is : NVIC_ICER2 = interruptSave[2]; in SPI.h

FrankBoesing, 2016-11-15:
Can look here: https://forum.pjrc.com/threads/36706-ILI9341-with-fullscreen-DMA-Buffer-for-Teensy-3-5-Teensy-3-6-only?p=123755&viewfull=1#post123755
Please...

FrankBoesing, 2016-11-16:
Provided a patch for the waveplayer.
But, still, this is a more general problem, as we have more than 1 SPI now, and more possibilibilities for SD-Cards. Anyway, i'm closing this.

closed 2016-11-16 by FrankBoesing

PaulStoffregen, 2016-11-23:
> Is there a way to give DMA priority for bus-accesses ?

Yes.  I believe it's possible to crossbar switch to favor one bus master over the others.  For K66, see chapter 20 starting on page 417.

Within the 16 or 32 DMA channels (all of them bus master number 2), it's also possible to program 2 levels of priority.  If you have DMA channels that are really urgent, they can be programmed at the higher priority level.  It only supports 2 priority levels, but that does allow high priority DMA channels to preempt (effective interrupt) the operation of the lower priority DMA channels.  See section 24.3.17 on page 552 for the priority setting.  The top 2 bits control which DMA channels can preempt the operation of others.