PaulStoffregen/cores #207: Print::print members have undefined behaviour

eric-wieser, 2017-01-19:
[These lines](https://github.com/PaulStoffregen/cores/blob/master/teensy/Print.h#L67-L85) all contain code of the form `return print() + println()`.

C++ makes no guarantee that `print()` will be invoked before `println()` here!

FrankBoesing, 2017-03-02:
Does anyone use an other compiler than GCC or a GCC derivate ? If yes, which ?

eric-wieser, 2017-03-02:
@FrankBoesing: Is that relevant? Does GCC make this guarantee, with optimization enabled?

FrankBoesing, 2017-03-02:
Can you post an example where the existing code does not work ?

PaulStoffregen, 2017-03-02:
I'm leaving this issue open, but unless there's a case where it matters, it's at the lowest priority.

eric-wieser, 2017-03-02:
@FrankBoesing: No, because the nature of undefined behaviour makes constructing a minimal example impossible. Any code you have that currently works could one day decide to stop working due to an unrelated change elsewhere in your code that causes the compiler to make a different optimization.

Note that the fix to this problem is trivial - implement this the same way as core Arduino does.

twilkers, 2017-03-02:
I have a code set that uses print variants to provide a text only webpage of calculations and have been chasing intermittent issues for months.  I'm wondering if this could be the source of my problem.  The symptom is that the image works well, but after some amount of time the application just stops servicing the webpage request.  A simple reboot clears the issue.  The amount of time can be from hours to a day or so.  The code set is not huge, and I have not been able to cut it down and get the same issue.  

Would you have a suggestion on how to determine if this is the source of my issue?

Thanks,
Tim

eric-wieser, 2017-03-02:
@twilkers: Worst case scenario, you'll see newlines in weird places. I guess that could be enough to corrupt a returned HTTP response, if you're building that manually too.

twilkers, 2017-03-02:
The page formatting seems consistent.  I just used the web server example in the tools and built it up from there.  

Thanks,
Tim