PaulStoffregen/cores #218: MIDI device name can only be set once

alex-Arc, 2017-02-14:
i have tested it on a T-LC and T3.2
i can only set the midi device name once.
compiling and uploading a new one even after restarting Arduino and the teensy loader results in the same name.

```
#include "usb_names.h"

// Edit these lines to create your own name.  The length must
// match the number of characters in your custom name.

#define MIDI_NAME   {'M','u','t','e'}
#define MIDI_NAME_LEN  4

// Do not change this part.  This exact format is required by USB.

struct usb_string_descriptor_struct usb_string_product_name = {
        2 + MIDI_NAME_LEN * 2,
        3,
        MIDI_NAME
};
```

Theremingenieur, 2018-01-24:
Depends on the OS. Some cache the device name, so that further changes aren’t displayed correctly. After changing the name again, plug the Teensy on a fresh computer and you’ll see the new name.

PaulStoffregen, 2018-01-24:
This example and the website have been updated with improved instructions.

Numerous people have confirmed this works.  I'm closing this old issue.

closed 2018-01-24 by PaulStoffregen