PaulStoffregen/cores #220: Keyboard.print(..) only supports ASCII while TeensyDuino Keyboard.press(..) supports ISO-8859-1

metachris, 2017-03-14:
Hello,

First off, thanks for your work on TeensyDuino!

We needed to simulate a Keyboard HID device and print ISO-8859-1 characters, but found the `Keyboard.print(..)` function only supports ASCII, while `Keyboard.press(..)` supports ISO-8859-1.

Is there a specific reason for this? Should TeensyDuino possibly override the default `Keyboard.print(..)` function to suport ISO-8859-1?

We solved it for our project with instead iterating over the string and calling `Keyboard.press` and `Keyboard.release` for each character.

PaulStoffregen, 2017-03-14:
Please try File > Examples > Teensy > USB_Keyboard > LayoutTest

ChrisVeigl, 2017-03-14:
I tried the LayoutTest example with a TeensyLC, German Keyboard Layout and USB-Type Settings "Keyboard" or "Serial + Mouse + Keyboard + Joystick"

here the results:

Teensy USB Keyboard Layout Test
Lowercase:  abcdefghijklmnopqrstuvwxyz
Uppercase:  ABCDEFGHIJKLMNOPQRSTUVWXYZ
Numbers:    0123456789
Symbols1:   !"#$%&'()*+,-./
Symbols2:   :;<=>?[\]^_`{|}~
Symbols3:  
Grave:     
Acute:     
Circumflex:
Tilde:     
Diaeresis: 
Cedilla:   
Ring Above:
AE:        
Thorn:     
Sharp S:   
O-Stroke:  
Eth:       
Euro:   

it seems that everything beyond "Symbols2" causes problems ...

I can confirm that calling Keyboard.press() and .release()  displays e.g. the Diaeresis characters öäü correctly (and prabably also the other characters)