PaulStoffregen/cores #226: USB serial: written bytes get read again, leading to exponential explosion

SimonSapin, 2017-03-21:
Hello. I made a project based on Teensy 3.2 a few month ago that ended up successful. Lately I wanted to start another project so I started with a simple "echo" program: have the Teensy wait until some input is available to read over serial USB, read one byte, write that same byte back over serial USB, and start over. On the other side (a Linux laptop) I used `cat /dev/ttyACM0` in a terminal and `echo "a" > /dev/ttyACM0` in another. The output was unexpected:

```
a

aa

aaaa

aaaaaaaa

aaaaaaaaaaaaaaaa

aaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa

```

After that the Teensy seemed hanged until I ran `teensy-loader-cli` again.

After some trial and error I ended up with a different program:

```c++
#include "WProgram.h"

extern "C" int main(void)
{
    int count = 1;
    while (count < 64) {
        int c = Serial.read();
        if (c >= 0) {
            Serial.write(c + 1);  // Note +1 here
            count++;
        }
    }
    Serial.write("\n");  // Necessary to get any ouptut. Maybe flushes some buffer?
    while (1) {}
}
```

This time, using `python -c "open('/dev/ttyACM0', 'wb').write(b'0')"` produces this very curious output:

```
112122312232334122323342334344512232334233434452334344534454556
```

With some line breaks added:

```
1
12
1223
12232334
1223233423343445
12232334233434452334344534454556
```

It looks like everything given to `Serial.write` is appended to some buffer which is never cleared, and `Serial.read` also copied bytes from that same buffer, going back to its start when reaching its end. So the size of this buffer doubles until something overflows (which seems to happen around 640 bytes).

Do you have any idea how this could happen? The runtime (code from this repository) is the same code that worked fine a few months ago. Maybe my Teensy got damaged in the meantime, but could a hardware fault cause a bug like this? The only (I think) other thing that changed is the compiler version, I now use arm-none-eabi-gcc 6.3.0-1 and arm-none-eabi-newlib 2.5.0.20170228-1. (With system packages from ArchLinux.)

SimonSapin, 2017-03-21:
@jamesmunns, have you seen anything like this?

jamesmunns, 2017-03-21:
Haven't seen it before, and a quick glance at the underlying souce doesn't show anything crazy. Do you get the behavior if you actually check if there is pending data? e.g.

```c++
#include "WProgram.h"

extern "C" int main(void)
{
    int count = 1;
    while (count < 64) {
        while(!Serial.available());
        int c = Serial.read();
        if (c >= 0) {
            Serial.write(c + 1);  // Note +1 here
            count++;
        }
    }
    Serial.write("\n");  // Necessary to get any ouptut. Maybe flushes some buffer?
    while (1) {}
}
```

jamesmunns, 2017-03-21:
I can try to repro later tonight, as well as try it with an Arduino Mega to see if the behavior is the same.

SimonSapin, 2017-03-21:
Unfortunately, adding `while(!Serial.available());` doesn’t change anything.

PaulStoffregen, 2017-03-21:
Please try using the Arduino Serial Monitor.  From the details here, it seems you're not running Arduino.  Please do so, even if you never wish to actually program with Arduino.  Just run it, select Teensy from Tools > Boards (do this first, since the serial monitor changes depending on whether Teensy is selected).  Then select /dev/ttyACM0 from Tools > Ports, and open the serial monitor window.

PaulStoffregen, 2017-03-21:
My guess is your custom code may be unwittingly using one of many unix terminal settings which processes the data in various ways.

Please, before you continue troubleshooting with new code, compare to what actually happens when you use the widely used Arduino Serial Monitor.  Even if you hate Java & Arduino, running this well known & widely used software for the sake of comparison and troubleshooting is essential.

SimonSapin, 2017-03-21:
I’ve managed to reproduce the issue on a different Teensy (though a LC instead of 3.2), so faulty microcontroller seems less likely.

PaulStoffregen, 2017-03-21:
Please try to reproduce the problem with the Arduino Serial Monitor.

SimonSapin, 2017-03-21:
@PaulStoffregen Thank you for this tip, it was very helpful. I can not reproduce the issue in the Serial Monitor, which shows that the issue is not on the Teensy side as I had assumed but on the Linux side. Better yet, now using the tty device as a file on the command line works fine too! The output of `stty -F /dev/ttyACM0` has changed.

Before (fortunately I still had this in an open terminal):

```
speed 9600 baud; line = 0;
-brkint -imaxbel
```

After:

```
speed 9600 baud; line = 0;
min = 100; time = 2;
-icrnl -imaxbel
-opost -onlcr
-isig -icanon -echo
```

I will study what these keywords mean and find out which ones are relevant.

By the way I laughed at the phrasing of your message :) It sounds like you may have dealt with many people who hate Arduino with a passion. I don’t particularly hate the Arduino IDE, it’s just not for me. I’m used to my text editor and terminal. And more importantly, @jamesmunns and I have been playing with using the Rust language for code running on ARM-based Teensies, which is probably completely unsupported by the Arduino IDE. What we have so far is at https://github.com/jamesmunns/teensy3-rs/

closed 2017-03-21 by SimonSapin