PaulStoffregen/cores #240: ADC Clock to set to fast for none T3.5/T3.6 devices.

Donziboy2, 2017-04-22:
Looking at the Analog.c file I noticed that several of the 8 bit speed settings are above(18MHz) what a T3.2 and below boards are rated to.  
Is this a mistake or is it safe to run them at these speeds?

[image]

PaulStoffregen, 2017-05-02:
good question....

pedvide, 2017-10-16:
The datasheet doesn't explicitly say that this can cause damage, only that the noise and resolution can't be guaranteed. 
Perhaps we could run some tests, maybe check the noise and resolution of a T3.2 at normal speed for a day or so, then repeat it for the the fastest speed, and check again at normal speeds. If the results at normal speeds are the same before and after then it should be fine.

Donziboy2, 2017-10-19:
pedvide, I also compared your clocks.  See below your pushing the speeds a little ;)
[image]

pedvide, 2017-10-19:
I already fixed those errors, thanks a lot! (I haven't added the missing values yet though.)
If you use "very high speed" then indeed it can be above the maximum rating, but if you use "high speed" then it's always within specs.

Also notice that the maximum speed for T LC, 3.0, 3.1/3.2, and 3.5 is 18 MHz (below 16 bits), but for T3.6 it's 24 MHz.

Donziboy2, 2017-10-19:
Cool, I will take a look and update my table.

I guess in the end its tough to say if it will just degrade the signal or cause actual damage.  I found out the hard way that using Pin13 as SCK while using DMA ADC was a disaster waiting to happen.  The trace for pin 13 on the T3.6 runs right under the micro from 1 side to the other, even with a 100R resistor in series it was adding 100mV pk-pk spikes onto my ADC readings.  Even when using alternate SCK pins it still looks awful(50mV pk-pk), I believe in the spec they say specs are accurate with other peripherals turned off/low power and I can see why.

pedvide, 2017-10-19:
Yes. Critically, the specs are only given for differential 16 bit mode. I assume that the traces of the differential pins follow parallel courses inside the chip, so that all those perturbations cancel out.