PaulStoffregen/cores #244: T3.5/3.6 ADC Ports Missing

Donziboy2, 2017-05-11:
ADC's 10, 11, 21, 22, 25 and 26 are listed but the Ports for those pins 64 - 69 are not listed in core_pins.h.  Aren't they needed?  Don't have a 3.6 to test it on atm.  I have found them on the schematic.

Cant seem to actually find 68/69 on the schematic....

pedvide, 2017-10-18:
Aren't those pins analog only? They are defined as CORE_ADCXX_PIN, see line 1284 and following.
Pins 68/69 (A25, A26) are D+, D- on the schematic. They only exist for Teensy 3.5, not for 3.6.

Donziboy2, 2017-10-19:
I see what you mean.  A10 and A11 are Diff pins with no other features.  A21 and A22 are the DAC's, but they also have ADC0_SE23 and ADC1_SE23 listed, so they can do both.  A25 and A26 are indeed 3.5 only :/

closed 2017-10-19 by Donziboy2