PaulStoffregen/cores #258: NeoGPS Compile  "pgm_read_ptr" redefined on T3.5

mjs513, 2017-10-10:
Using IDE 1.8.5 and TD1.40 I tried compiling a program I have using NeoGPS and I received the following warning message:
```
Programs\arduino-1.8.5\hardware\teensy\avr\cores\teensy3/WString.h:33:0,
                 from C:\Local Programs\arduino-1.8.5\hardware\teensy\avr\cores\teensy3/Print.h:38,
                 from C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS\src\DMS.cpp:20:
C:\Local Programs\arduino-1.8.5\hardware\teensy\avr\cores\teensy3/avr/pgmspace.h:113:0: warning: "pgm_read_ptr" redefined
 #define pgm_read_ptr(addr) ({ \
 ^

In file included from C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS\src\DMS.h:21:0,
                 from C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS\src\DMS.cpp:18:
C:\Users\CyberPalin\Documents\Arduino\libraries\NeoGPS\src\NeoGPS_cfg.h:106:0: note: this is the location of the previous definition
   #define pgm_read_ptr(addr) (*(const void **)(addr))
 ^
```

I ran an example neogps sketch (tabular.ino) to verify that I received the same error message for the T3.5 and I did.  I then compiled it for the Arduino Uno and no warning.  Looking at the pgmspace.h file for Arduino believe that all that needs to be added to the Teensy3 pgmspace.h file is the missing definition for pgm_read_ptr:
   #define pgm_read_ptr(addr) (*(const void **)(addr))

Thanks
Mike

PaulStoffregen, 2017-10-11:
This is an issue with NeoGPS, not Teensy.  Please report it to NeoGPS.

    //------------------------------------------------------------------------
    // The PROGMEM definitions are not correct for Zero and MKR1000
    
    #if !defined(__AVR__)
      // TODO: use the Zero/MKR1000-specific symbols
      #undef pgm_read_ptr
      #define pgm_read_ptr(addr) (*(const void **)(addr))
    #endif

closed 2017-10-11 by PaulStoffregen

mjs513, 2017-10-11:
Done.  Thank you.