PaulStoffregen/cores #264: Issue sending sysex with usbMIDI

neroroxxx, 2018-01-15:
There seems to be an issue with usbMIDI, i'm testing with multiple t32s, anytime i send a system exclusive 3 messages are sent, always in the same order

#1 [0xF0,0xF7]
#2 the actual array i wanted to send
#3 [0x00,0xF7]

This seems to happen no matter what. below is the sketch i'm using to test it, i'm using teensyduino 1.41 beta3 with arduino 1.9.0-beta31 using the lastest commit of usb_midi.h and usb_midi.c

const uint16_t size = 10;
uint8_t data[size] = {0xF0,1,2,3,4,5,6,7,8,0xF7};
void setup(){

}
void loop(){
  usbMIDI.sendSysEx(size,data);
  delay(2000);
}

PaulStoffregen, 2018-01-16:
You need to use:

    usbMIDI.sendSysEx(size, data, true);

The 3rd parameter defaults to false, meaning your buffer does not have the 0xF0 and 0xF7.

Or you could use this:

    const uint16_t size = 8;
    uint8_t data[size] = {1, 2, 3, 4, 5, 6, 7, 8};
    void setup() {
    }
    void loop() {
      usbMIDI.sendSysEx(size, data, false);
      delay(2000);
    }

neroroxxx, 2018-01-16:
ok thank you, was that just added? I know the MIDI library had it but i don't remember your usb midi libraries having them, eitherway thank you!

closed 2018-01-21 by PaulStoffregen