PaulStoffregen/cores #270: IPAddress.h should also define operator!=()

ssilverman, 2018-02-20:
It is my understanding that if operator== is defined, then operator!= must be also.

PaulStoffregen, 2018-03-16:
IPAddress.h is for compatibility with Arduino.  They don't have this in their IPAddress.h.

ssilverman, 2018-03-16:
<frown>
So this means, in theory, that when comparing two IPAddresses, !(a == b) must be used instead of (a != b)? At least that’s what I’m getting out of this. Yikes. I’ll go file a bug report with them.

PaulStoffregen, 2018-03-16:
Yes, please ask Arduino to fix this in their IPAddress.h.  Then I will adopt whatever fix they choose.

I hope you can understand why I'm reluctant to allow this sort of compatibility code to diverge from Arduino's version.

ssilverman, 2018-03-16:
Completely. Lots of Arduino code out there you’d want to be compatible.

ssilverman, 2018-03-16:
I hit ‘Close’ by mistake. I’ll go file this with them and update here...

ssilverman, 2018-03-16:
I just submitted https://github.com/arduino/Arduino/issues/7336.

ssilverman, 2018-11-12:
Looks like they changed it. :)

ssilverman, 2019-12-17:
For posterity, here's where they changed it: https://github.com/arduino/ArduinoCore-API/commit/d5ae3faba1d69500f920c57b2afac59021831278
(Issue https://github.com/arduino/Arduino/issues/7336)

PR #612 (2022-04-23): Add IPAddress::operator!=()
Fixes #270.

See changes to ArduinoCore-API:
* https://github.com/arduino/Arduino/issues/7336
* https://github.com/arduino/ArduinoCore-API/commit/d5ae3faba1d69500f920c57b2afac59021831278
closed 2022-04-23 by PaulStoffregen