PaulStoffregen/cores #291: [Feature Request] Add the ability to pass state to the function called by IntervalTimer

ssilverman, 2018-11-13:
Please use this Github issue *only for reporting a specific problem*.

Please **do not use Github issues for project troubleshooting.**  Post on the PJRC Forum for Teensy-related troubleshooting.

https://forum.pjrc.com/forum.php

General suggestions for new features should also be discussed on the forum.  Github issues for new features are usually only appropriate after decisions are made via forum discussion.

Also check out the most common troubleshooting page:

https://www.pjrc.com/teensy/troubleshoot.html

Before reporting any issue, please make sure you are using the latest Teensyduino version.

https://www.pjrc.com/teensy/td_download.html

If a beta test version is available, please check whether it fixes the issue.

When reporting any issue, please try to provide complete information.  The following are almost always needed:

- A complete copy of the code (which can be copied into Arduino and run on a Teensy board) which reproduces the problem is usually needed to investigate any issue.  Please try to trim your program to the minimum necessary, ideally not depending on extra hardware if possible.  But a complete program that can be copied into Arduino without guesswork is essential.  Please do not submit this Github issue with only code fragments or a vague description of code to recreate the issue.

- Which Teensy board(s) are you using.

- If other hardware is needed to reproduce the issue, please provide links or part numbers where this hardware can be purchased.  Very specific details showing how the hardware is connected also help.  Post photos if needed.

- Which version of Teensyduino and Arduino were used, what Tools menu settings were chosen, and which operating system was used.  If other software is needed, please be specific.

PaulStoffregen, 2018-11-13:
std::function is under consideration.  But no changes are likely until well after the next generation of Teensy hardware is shipping, probably mid-2019.

ssilverman, 2018-11-21:
Many of the Arduino-style libraries assume global state, but larger programs often aren't written this way. Not passing any state to callbacks precludes this non-global-style of programming, even without the use of srd::function. I'm going to see what I can do with a fork of IntervalTimer...

Basically, I'm going to add methods that looks like this:
`begin(void (*funct)(void *state), void *state, some_type micros)`

Doing it this way means that it's possible to stop the timer that called it, for one-shot use, and also to operate on other relevant state—without using any global variables.

ssilverman, 2018-11-21:
Here's something like I was thinking:
https://github.com/ssilverman/cores/commit/6051e61387baa9a03828e8bd178335480bdc7757

PaulStoffregen, 2018-11-27:
Considering std::function for this, attachInterrupt and other APIs.  But right now focused on hardware, so not planning to make this sort of change (or any change requiring much testing and possible compatibility issues) until middle of 2019.

jbliesener, 2018-11-27:
@PaulStoffregen, you may consider reusing the approach I built into the FlightSim USB library. Of course, std::function would be cleaner and more convenient, but the void* context data used with FlightSimInteger/FlightSimFloat works very well.

PaulStoffregen, 2018-11-27:
Sorry, I can't do this sort of change now, while focusing on hardware.

Let's talk of this next year, after the new hardware is released.

ssilverman, 2019-11-27:
I swear it's a coincidence that this is almost exactly a year since the last post here. :)

I was wondering if revisiting this issue is possible? This way I don't have to use my own IntervalTimer implementation. What can I do to help?

ssilverman, 2019-11-27:
Here's a sample commit: ssilverman/cores@8024283d9f9066d5293f2b69a17d3cf1e1d6ad56.

ssilverman, 2020-09-12:
I'd love to explore getting some sort of state-passing change into `IntervalTimer`. I'm finding that my https://github.com/ssilverman/TeensyDMX library conflicts with https://github.com/vjmuzik/NativeEthernet, likely because of how I use the PIT timers from my own `PeriodicTimer` class. I'd love to use the same `IntervalTimer` class, but it's difficult/messy to do things like one-shot timers, where a callback needs to know what/who called it.

luni64, 2020-09-12:
Actually it is quite simple to pass constant state like the address of the timer to the callback. E.g. the code below implements a one-shot timer.

```c++
void cb(IntervalTimer* caller)
{
    Serial.println(millis());
    caller->end();
}

IntervalTimer timer;

void setup()
{
    timer.begin([] { cb(&timer); }, 1'000'000);  // use a lambda expression as relay to invoke the actual callback
}

void loop()
{
}
```
If you need more elaborate things like embedding the timer to your calls and use class members as callbacks you can have a look here:  https://forum.pjrc.com/threads/63003-IntervalTimerEx-Callbacks-with-state

ssilverman, 2020-09-12:
That solution uses a global `cb` function, which is my problem to begin with; I need non-global things. Your IntervalTimerEx project looks promising, thanks for the pointer.