PaulStoffregen/cores #293: Invalid string descriptor index

mbenkmann, 2018-12-13:
https://github.com/PaulStoffregen/cores/blob/44359dcf6b3d11fc59f8c386e518db7f1f2a5ad1/teensy3/usb_desc.c#L606

Unless I'm misunderstanding something this seems to refer to a string descriptor with index 4 which does not exist by default.

pierstitus, 2019-12-18:
The same on line 1104, and in teensy4/usb_desc.c on lines 624, 1131, 1458 and 1965.

lusb -v reports `(error)`, for example:

    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass          2 Communications
      bFunctionSubClass       2 Abstract (modem)
      bFunctionProtocol       1 AT-commands (v.25ter)
      iFunction               4 (error)

On linux it just tries again or neglects the error. On a stm32 as usb host the enumeration halts on such errors.

So either change all those 4's to 0's, or create a real string for them.

Thanks for the great work!

closed 2020-01-07 by PaulStoffregen

pierstitus, 2020-01-08:
Thanks for fixing, any reason you left the MTP descriptor strings out?
teensy3/usb_desc.c line 624 
teensy4/usb_desc.c lines 1131 and 1965

PaulStoffregen, 2020-01-08:
Opps, missed those.  Been focusing on the Wire lib.

Fixed.  https://github.com/PaulStoffregen/cores/commit/d4d59c98b7b2461f24f5c6a948910ecd811f8157

commit d9fe2d70 (2020-01-07): Don't reference non-existant usb string descriptor
fixes #293, fixes #414