PaulStoffregen/cores #388: Teensy35 Stream.timedRead() is protected

paulhayes, 2019-08-21:
This method is called by the following library: ArduinoHttpClient @ src\HttpClient.cpp:576:17

The method is public in the main arduino library and other compatables, shouldn't the Stream class be symmetrical across the different platforms? Could you modify this to match arduino framework?

ArduinoHttpClient is listed as being compatible with Teensy, and it is except for this one methods protection level.

FrankBoesing, 2020-12-06:
It is protected here: https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/Stream.h#L54 too

closed 2020-12-13 by paulhayes

herrfrei, 2021-01-07:
I have the same problem, but in the code for Teensy3 and Teensy4 the method in declared private, not protected! Hence the different HttpClient libraries wont't compile.