PaulStoffregen/cores #409: Should TFLG in IntervalTimer (teensy4) be cleared even if the function is nullptr?

ssilverman, 2019-12-16:
https://github.com/PaulStoffregen/cores/blob/06492c1d11b93879b74d8db586d4acce5f2ea5ea/teensy4/IntervalTimer.cpp#L108

It looks like if the function is `nullptr` then the interrupt won't get cleared. Should the interrupt be cleared anyway?

FrankBoesing, 2020-04-06:
That's correct - on the other hand, i've not heard of any problem, and a fix would be slower.

ssilverman, 2021-09-15:
There's two checks anyway, would a fix really slow this down?

PaulStoffregen, 2023-01-17:
Has anyone actually encountered this issue in practice?

This change could have a performance impact.  The common case of an unused PIT channel causes 1 bus access to fast DTCM.  The rest is skipped if the pointer in fast DTCM is null.  #610 changes the common case to reading a peripheral register which is done over slower AXI bus and peripheral bridge.

I'm not planning to merge #610 unless a test case or user report shows it can result in wrong behavior in practice (using the IntervalTimer public API) or if the performance impact is carefully analyzed / benchmarked.

ssilverman, 2023-01-17:
Understood. That’s a very interesting reason. (Today I learned.) :)