PaulStoffregen/cores #412: Add support for Arduino IDE 3rd party hardware specification

Legion2, 2020-01-04:
Use the Arduino IDE 3rd party hardware specification to integrate the teensy boards into Arduino IDE without the need to install an add-on for Arduino IDE. I don't understand why the Teensyduino add-on is required, what feature does it support that is not supported by native Arduino IDE with the 3rd party hardware specification?

I'm an Arduino developer and use multiple different boards with different architecture and manufacturer in Arduino IDE without any problem and don't want to install Teensyduino and mess up the other boards.

PaulStoffregen, 2020-01-04:
Most of the patches are to the serial monitor.

Legion2, 2020-01-05:
OK, but the Serial Monitor is for humans, who can process 20 Mbyte text input per second? This data transmits makes only sense when used by another program on the host system.

That looks like a nice feature to me, but the serial monitor is not required for normal operation of the IDE as compiling and uploading sketches. And even if I need a serial monitor for debugging etc. 1 Mbit/sec is enough.

So would it be possible to support the teensy boards in Arduino IDE without the modified Serial Monitor and only with the Arduino 3rd party hardware definition? The modified Serial Monitor is then a optional feature that can be installed if needed.

PaulStoffregen, 2020-01-05:
This really isn't up for debate, at least not here on a github issue.  If you want to talk further, please do so on the forum.

closed 2020-01-05 by PaulStoffregen

PaulStoffregen, 2022-08-31:
Quick followup on this old issue.  Much has changed in the last ~2.5 years.  Arduino expanded their 3rd party hardware specs with pluggable discovery and pluggable monitor, which Teensy now supports for the upcoming Arduino IDE 2.0.

https://www.pjrc.com/teensyduino-1-57-released/

Arduino isn't planning to update the java-based 1.8.x IDE, so the installer will continue to be necessary for pre-2.0 versions.  But for the new IDE they're building, a special installer isn't needed.

Legion2, 2022-09-03:
Thanks for the information, will look into it and try to get my library running on teensy boards.