PaulStoffregen/cores #423: defines for Random Number Generator Accelerator on Teensy 4.0

phulst, 2020-01-27:
It appears that the necessary definitions to use the RNGA on Teensy 4.0 are missing.  
I followed example code from

https://forum.pjrc.com/threads/48745-Teensy-3-6-Random-Number-Generator

but this doesn't compile for Teensy 4. I found that the define for RNG_CR is missing. This is present in the kinetis.h file for teensy3 but I found no such definition in the teensy4 definitions.

phulst, 2020-01-28:
after digging  around  in the forums a bit more I did  come across some sample code  for the  T4. 

https://forum.pjrc.com/threads/54711-Teensy-4-0-First-Beta-Test?p=195000&viewfull=1#post195000

that will likely unblock me for now. However it would still be nice if the T4 TRNG registers could be included with the core libraries.  I would send you a pull request if I was more well-versed with this stuff and knew all the best practices for embedded C code...

PaulStoffregen, 2022-08-31:
Fixed by https://github.com/PaulStoffregen/cores/commit/c9e3cc0daecddb3a28f28e7053db63f94ffd40dd

PaulStoffregen, 2022-08-31:
If anyone finds this old issue, code was added to the Entropy library to provide TRNG access.

closed 2022-08-31 by PaulStoffregen