PaulStoffregen/cores #429: AVR register emulation on Teensy 4.0 GPIO[ABCD]_* not existing

imwhocodes, 2020-02-04:
Hi
I'm using PlatformIO (4.1.0) with the latest Teensyduino Version (1.49)

Working on a pulse/step generator and I'm using a combination of DMA and BitBand GPIO register (_**GPIOD_PTOR, GPIOD_PDOR** and similar as DMA target address_) 

Everything is working fine on Teensy3.x, but when I tried on the new Teensy4.0 the mentioned **#define** values weren't found

I was supposing that the changelog line saying  "AVR register emulation on Teensy 4.0" meant those mentioned set of macros 

If is not so: is there any way to achieve the same result on teensy4.0 (toggling PINs using DMA, parallels to CPU execution)?

Are this macros (https://github.com/PaulStoffregen/cores/blob/master/teensy4/imxrt.h#L5235) the new/proper ways to address such PINs memory area? Do they point to the right address or are they only for configuring interrupts?

Thanks,
Luca

WestfW, 2020-02-05:
Note that cortex-m7 does not support bit-Banding
https://community.arm.com/developer/ip-products/processors/f/cortex-m-forum/10266/why-cortex-m7-doesn-t-support-bit-banding

PaulStoffregen, 2020-02-05:
Generally digitalWrite() and digitalWriteFast() are the best approach.

Direct register access has been discussed many times on the forum.  Here are some of those threads.

https://forum.pjrc.com/threads/58377-Reading-multiple-GPIO-pins-on-the-Teensy-4-0-quot-atomically-quot

https://forum.pjrc.com/threads/58405-Teensy-4-0-has-no-PORTC-and-PORTD-definitions

https://forum.pjrc.com/threads/57972-Raft-of-Teensy-4-pin-port-parallel-read-questions

https://forum.pjrc.com/threads/57738-DMA-can-t-write-to-GPIO-registers-on-teensy-4-0

https://forum.pjrc.com/threads/57717-port-and-toggle-question-T4

https://forum.pjrc.com/threads/57698-Parallel-IO-is-it-possible

If none of those answer your question, please ask on the forum.  This issue tracker is only for reporting software defects.  It's not the place to ask questions.  Use the forum for questions.

closed 2020-02-05 by PaulStoffregen