PaulStoffregen/cores #441: T4: WFI: Interrupts do not wake core.

FrankBoesing, 2020-02-24:
void loop {
Serial.println("ZZzzzzz");
delay(1);
asm ("wfi");
Serial.println("Good Morning");
delay(100);
}
At minimum the systick should wake up the T4 - however the program stops after "ZZzzzz"

I guess the idel mode needs to be configured somehow.

FrankBoesing, 2020-02-24:
Maybe this PDF is useful: https://www.nxp.com/docs/en/application-note/AN12085.pdf

FrankBoesing, 2020-02-24:
Ok... WFI DOES wake the core :) It's just that it happens not in short intervals with systick (other than on T3.x)
If you send a text in SERMON , the T4 wakes up.

Ok, I can live with that. It's just different on the 3.x
Question is, wether this needs a fix or not... i don't know.

Defragster, 2020-03-03:
I've seen inconsistent results from use of WFI on T_4.

If a Serial.print() is outbound it wakes?  If not it never wakes was the last I 'inferred' when I left that sketch.

There is an NXP_DOC that explains 'proper and complete use of WFI' on the IMXRT - I linked it during T4_Beta and got no mention.  It is an involved process it seems of setting things right to do WFI and then to return from WFI.

https://www.nxp.com/docs/en/application-note/AN12085.pdf

System Idle is the lowest 'idle' mode on WFI ...

[image]

FrankBoesing, 2020-03-03:
For me, it'ok as it is. Better than on T3.x where systick wakes it.
Deeper sleeping can be handled by a library.

closed 2020-03-07 by FrankBoesing