PaulStoffregen/cores #449: Keeper -  Issues with analog

FrankBoesing, 2020-03-12:
The RM says:

> The ADC input signals connect to GPIO. The GPIO default
> configuration is enabled for keeper. The keeper causes an
> undesired jump behavior in ADC. To avoid the problem,
> disable keeper before starting ADC. For detailed information
> about keeper, refer to the GPIO block.

I'd say we should disable the keeper for all pins in startup.c
https://forum.pjrc.com/threads/60005-Teensy-4-0-ADC-keeper-issue

ossiAachen, 2020-03-13:
I think I now understand why the keepers are enabled after reset: This brings the inputs in a definite state and avoids undefined voltages on the inputs. In order to use the ADC on a pin you have to set it to "input" by using pinmode().

Theremingenieur, 2020-03-13:
The keeper has its reason to exist, thus a systematic disabling is probably not the optimal way, especially as not all pins are forcibly used for ADC readings in an average project. I'd rather suggest disabling it individually in the analogRead() code and reenable it afterwards. Similar additions with a #if defined() for the Teensy 4.x should be made in the ADC library.

FrankBoesing, 2020-03-13:
So what do you Think is the reason, if its not the one mentioned in the appnote?

FrankBoesing, 2020-03-13:
The keeper is incompatible to all Teensy models without it.
If a user wants it,  he can enable it for this special case.

ossiAachen, 2020-03-15:
@Theremingenieur:
I think disabling keeper only for the duration of a analogRead() is not a good idea, because it will lead to problems if a large DC-blockin capacitor is used in connecting the signal to th DC, a practice that is often chosen. I would like o have keepers off for all time and only switched on by the user.

FrankBoesing, 2020-03-15:
The pins are pulled low by default (without pinMode) which is incompatible to any other Arduino. So for me it's out of question that it _must_ be changed.

Theremingenieur, 2020-03-15:
NXP did build the keepers for a reason. Ok, my opinion doesn’t obviously count here, thus make what you judge best. But for me, it’s not understandable why one would sacrifice new features to maintain backwards-compatibility. What about new projects which don’t use these degraded (compared to the Teensy 3.x) ADCs of the T4.x? If you want to disable the keepers by default, please add at least a function which allows to reenable these.

FrankBoesing, 2020-03-15:
This is just a discussion, all opinions count :)
So, NXP had a reson not to add keepers to the other chips, too?

FrankBoesing, 2020-03-15:
You can read NXPs reason in the mentioned Application note.
Freel free to add a pullrequest.

Theremingenieur, 2020-03-15:
Thinking about extending the pinMode() parameters. What about adding OUTPUT_KEEPER To the list?

FrankBoesing, 2020-03-15:
As this is needed for input-mode anyway, i've added it to digitalWrite() 
https://github.com/PaulStoffregen/cores/pull/454

So this helps for the keepers, too: Just select inputMode and do a digitalWrite() - or use inputMode with PULLUP/PULLDOWN

Note: DigitalWrite would be a todo.. but this is no Arduino-function.

ossiAachen, 2020-03-16:
"You can read NXPs reason in the mentioned Application note.
Freel free to add a pullrequest."

Wich AppNote is meant and where can I get it?

FrankBoesing, 2020-03-16:
https://www.google.com/url?sa=t&source=web&rct=j&url=https://www.nxp.com/docs/en/application-note/AN5078.pdf&ved=2ahUKEwjxvbPOzp7oAhVR26QKHZH3B6AQFjAAegQIARAB&usg=AOvVaw2nPasH_5x-7bziVYCwG_ke

FrankBoesing, 2020-03-16:
It is for power savings - if you disable the output stage it can keep the state of the pin with pullup/pulldown resistors.

closed 2020-05-20 by FrankBoesing