PaulStoffregen/cores #464: problems setting teensy 4 RTC

manitou48, 2020-04-12:
Two problems setting RTC on T4.
1) bootloader is setting T4 RTC, but it is one hour behind PC's time?

2) set_rtc() in T4 rtc.c is not setting the proper T4 hardware RTC register for VBAT retention. see
https://forum.pjrc.com/threads/45377-TimeTeensy3-example-gt-Time-chosen-is-not-permanent?p=235422&viewfull=1#post235422

blackketter, 2020-04-29:
I can confirm that the fix in that thread resolves the VBAT issue
```
void rtc_set(unsigned long t)
{
   // stop the RTC
   SNVS_HPCR &= ~(SNVS_HPCR_RTC_EN | SNVS_HPCR_HP_TS);
   while (SNVS_HPCR & SNVS_HPCR_RTC_EN); // wait
   // stop the SRTC
   SNVS_LPCR &= ~SNVS_LPCR_SRTC_ENV;
   while (SNVS_LPCR & SNVS_LPCR_SRTC_ENV); // wait
   // set the SRTC
   SNVS_LPSRTCLR = t << 15;
   SNVS_LPSRTCMR = t >> 17;
   // start the SRTC
   SNVS_LPCR |= SNVS_LPCR_SRTC_ENV;
   while (!(SNVS_LPCR & SNVS_LPCR_SRTC_ENV)); // wait
   // start the RTC and sync it to the SRTC
   SNVS_HPCR |= SNVS_HPCR_RTC_EN | SNVS_HPCR_HP_TS;
}
```

PaulStoffregen, 2020-05-01:
oh, opps, this should probably credit Silverlock

closed 2020-05-01 by PaulStoffregen

bleckers, 2020-06-14:
I just wanted to add that it might be worth having both functions (as like a set/store). While using libraries like Smartmatrix, which are heavily timing dependent. The (new) rtc_set function causes display glitching with this library, as the device waits for the register to update and seems to block all DMA requests for this time (not the while loop in the set function, but something to do with setting SNVS_LPCR).

If this change is not desirable, then for others with similar issues, calling the old function when updating time just updates the RTC:

void rtc_volatile_set(unsigned long t)
{
  SNVS_HPCR &= ~SNVS_HPCR_RTC_EN;
	while (SNVS_HPCR & SNVS_HPCR_RTC_EN) ; // wait
	SNVS_HPRTCLR = t << 15;
	SNVS_HPRTCMR = t >> 17;
	SNVS_HPCR |= SNVS_HPCR_RTC_EN;
}

Then you can call rtc_set to update the SRTC when it is opportunistic to do so (during display sleep functions for example).

commit 536aa0ae (2020-05-01): Fix rtc_set on Teensy 4.0 (thanks blackketter)
fixes #464