PaulStoffregen/cores #476: Debug Enable - first Request from 2014

FrankBoesing, 2020-05-21:
https://forum.pjrc.com/threads/26358-Software-Debugger-Stack?p=52548&viewfull=1#post52548

It would be nice if you keep your promise of 2014 to disable  C_DEBUGEN

Thank you, 
Frank.

PaulStoffregen, 2020-05-21:
Wow Frank, your question sounds kind of hostile.  Are we still friends?

FWIW, on Teensy 3.x this promise was kept.  But it was never documented well (probably mentioned on the forum somewhere).  The way to disable C_DEBUGEN involves entering a low power sleep mode.  When the bootloader sees the main processor is sleeping, it will disable C_DEBUGEN and relinquish all control, then the bootloader chip goes into a deep sleep mode.

The MKL02 chip remains in a deep sleep mode until the pushbutton is pressed, so automatic uploading isn't possible.   Once in this mode, the only way to get the bootloader back is pressing the button.  But while it is disabled and sleeping, C_DEBUGEN is cleared and the bootloader will not make any attempt to access the main processor.

PaulStoffregen, 2020-05-21:
On Teensy 4.x, the bootloader does not yet detect IMXRT sleep modes, nor does it ever sleep.  I am planning to implement this in a future version.  But it is not so simple.  NXP did not put the "MDM" debug module inside the IMXRT chip.  On Teensy 3.x, the bootloader queries the MDM to detect the sleep modes.

FrankBoesing, 2020-05-21:
Certainly. We may not always agree, but we don't have to agree.

Thank you for pointing that out - I was not aware of it, or I read it
briefly and forgot about it.

Is there a short code snippet for it?

Maybe we can manage to implement software debugging after all.

FrankBoesing, 2020-05-21:
Ok, thanks.

closed 2020-05-21 by FrankBoesing