PaulStoffregen/cores #485: Add printf() format argument checking to Print class

bperrybap, 2020-08-16:
If you add a format attribute to the printf() declarations in the Print class, gcc will sanity check the arguments against format strings.

Here are the new declarations for Print.h

```
	int printf(const char *format, ...) __attribute__ ((format (printf, 2, 3)));
	int printf(const __FlashStringHelper *format, ...) __attribute__ ((format (printf, 2, 3)));
```

Changing these declarations is all that is necessary.

PaulStoffregen, 2020-10-22:
It only seems to work on the first case.  When I tried it on the 2nd, I get errors that the input is not a string type.

Another minor issue is uint32_t is defined as unsigned long, so "%ld" is expected.  Probably going to generate a lot of needless warnings.  Will give it a try for 1.54-beta1 anyway.

closed 2020-10-22 by PaulStoffregen

PaulStoffregen, 2020-12-01:
Late followup on this old issue.  Format checking on Serial.printf() was removed by this commit:  https://github.com/PaulStoffregen/cores/commit/ba177b0d22d8974b43c65753a710e54a4979eb7c

Turns out these warnings are far too pedantic.  Most of the pain revolves around warnings for "%d" and "%u" went the variable is uint32_t type.  It demands "%ld" or "%lu", because uint32_t is defined as long type.

Discussion here:
https://forum.pjrc.com/threads/62473?p=256873&viewfull=1#post256873

Will reconsider this when / if a future toolchain offfers a less pedantic warning where all 32 bit integers can be accepted for "%d".

commit 0bd7c42a (2020-10-22): Enable Serial.printf format warnings, fixes #485