PaulStoffregen/cores #499: Teensy 4: attachInterrupt configuration invalid for certain pins

ardnew, 2020-11-04:
In function `attachInterrupt` from Teensy 4 source file `cores/teensy4/interrupt.c` (below), the code tries to select the appropriate `ICR1` or `ICR2` register based on the bitmask of a given pin. These 32-bit registers each contain 2-bit interrupt configurations for 16 pins from a given GPIO port; i.e., `ICR1` configures pins 0-15, and `ICR2` configures pins 16-31.

However, the condition `if (index < 15) ...` incorrectly handles pins 0-14 with `ICR1` and pins 15-31 with `ICR2`. You'll see this is incorrect with the calculation `uint32_t shift = (index - 16) * 2;` when entering the else-block when `index` equals 15, resulting in `uint32_t shift = (15 - 16) * 2;`.

This condition should be changed to either `if (index <= 15) ...` or `if (index < 16) ...`. 

This bug affects all GPIO ports (`GPIO6`-`GPIO9`).

```c++
void attachInterrupt(uint8_t pin, void (*function)(void), int mode)
{
	if (pin >= CORE_NUM_DIGITAL) return;
	//printf("attachInterrupt, pin=%u\n", pin);
	volatile uint32_t *gpio = portOutputRegister(pin);
	volatile uint32_t *mux = portConfigRegister(pin);
	//volatile uint32_t *pad = portControlRegister(pin);
	uint32_t mask = digitalPinToBitMask(pin);

	voidFuncPtr *table;

#if defined(__IMXRT1062__)

	switch((uint32_t)gpio) {
		case (uint32_t)&GPIO6_DR:
			table = isr_table_gpio1;
			break;
		case (uint32_t)&GPIO7_DR:
			table = isr_table_gpio2;
			break;
		case (uint32_t)&GPIO8_DR:
			table = isr_table_gpio3;
			break;
		case (uint32_t)&GPIO9_DR:
			table = isr_table_gpio4;
			break;
		default:
			return;
	}

	attachInterruptVector(IRQ_GPIO6789, &irq_gpio6789);
	NVIC_ENABLE_IRQ(IRQ_GPIO6789);

#endif

	uint32_t icr;
	switch (mode) {
		case CHANGE:  icr = 0; break;
		case RISING:  icr = 2; break;
		case FALLING: icr = 3; break;
		case LOW:     icr = 0; break;
		case HIGH:    icr = 1; break;
		default: return;
	}

	// TODO: global interrupt disable to protect these read-modify-write accesses?
	gpio[IMR] &= ~mask;	// disable interrupt
	*mux = 5;		// pin is GPIO
	gpio[GDIR] &= ~mask;	// pin to input mode
	uint32_t index = __builtin_ctz(mask);
	table[index] = function;
	if (mode == CHANGE) {
		gpio[EDGE] |= mask;
	} else {
		gpio[EDGE] &= ~mask;
		if (index < 15) {
			uint32_t shift = index * 2;
			gpio[ICR1] = (gpio[ICR1] & ~(3 << shift)) | (icr << shift);
		} else {
			uint32_t shift = (index - 16) * 2;
			gpio[ICR2] = (gpio[ICR2] & ~(3 << shift)) | (icr << shift);
		}
	}
	gpio[ISR] = mask;  // clear any prior pending interrupt
	gpio[IMR] |= mask; // enable interrupt
}
```

FrankBoesing, 2020-12-06:
Yes, I think you're right.

FrankBoesing, 2020-12-06:
I'd say this can be closed, now :)