PaulStoffregen/cores #501: Small "bug" with pgmspace.h macros

drewhamiltonasdf, 2020-11-08:
Hi Paul,
I'm working on a port of some AVR code for ROS (Robot Operating System) that got hung up on your pgmspace.h macro hacks. You might already be aware of this already, but it doesn't seem to work if the original code returns a value with PSTR, because you replace PSTR(str) with two lines, and you can't return two lines of code...

```
#define PSTR(str) ({static const char data[] PROGMEM = (str); &data[0];})
```

Here's the exact code I got hung up on. It's trying to return a hard-coded checksum key that it's storing in flash:
```
const char * getMD5(){ return PSTR( "992ce8a1687cec8c8bd883ec73ca41d1" ); };
```

My hack for my project was to do something along these lines:
```
    const char * getMD5(){ 
	static const char data[] = "992ce8a1687cec8c8bd883ec73ca41d1";
    	return &data[0];
    };
```

Like I said, pretty esoteric, but maybe you can think of a clever workaround. If not, I just like to post solutions that are searchable for other folks that come along. I'm only trying to be helpful, and I very much appreciate all the incredible stuff you've put out into the world over the last years. 

Best,
Drew Hamilton

WestfW, 2020-11-08:
The original AVR code has:
`# define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))`

Does the __extension__ make the difference?

drewhamiltonasdf, 2020-11-08:
> The original AVR code has:
> `# define PSTR(s) (__extension__({static const char __c[] PROGMEM = (s); &__c[0];}))`
> 
> Does the **extension** make the difference?

Oh it is? Hmmmm...

PaulStoffregen, 2020-11-09:
@drewhamiltonasdf - A couple quick questions... Which Teensy are you using?  Are you using Arduino 1.8.13 with Teensyduino 1.53 to compile, or some other software?  Is this problem reproducible with a very minimal program which just prints from loop() without any other code?

Any chance I can talk you into saving me a little time with a trivial but complete program I can just copy into Arduino and upload to a specific Teensy model to reproduce the issue?

PaulStoffregen, 2020-11-09:
Here is the reason why I ask for a complete program....

[image]

As you can see, I copied the code you said doesn't work into Arduino, filled in setup() and loop() with minimal code and uploaded onto the Teensy I have on my workbench (which happens to be Teensy 4.1)... and I can't reproduce the problem!

Please, give me a complete program I can copy into Arduino and tell me which Teensy to use, so I can reproduce the problem without wasting any more time.

drewhamiltonasdf, 2020-11-09:
That's interesting. I was using a Teensy 4.1 with Arduino 1.8.13 and Teensyduino loader 1.53 in Ubuntu Linux. I ran the same short program you wrote to reproduce the issue, and it is not giving me any errors either.

If this short program works then my hypothesis fails. There must be something else going on with the code that was causing it to fail other than what I thought. I'll dig around and see if I can reproduce it with a smaller piece of code, but I think it is appropriate to close the issue.

PaulStoffregen, 2020-11-09:
Ok, closing this.  Please reopen if you believe it's really a bug in Teensyduino's code, and you can give me a complete program to reproduce it.

closed 2020-11-09 by PaulStoffregen