PaulStoffregen/cores #560: Missing ARDUINO_ARCH_{architecture} definition as described in the Arduino Platform Specification

lnlp, 2021-05-20:
The [Arduino Platform Specification](https://arduino.github.io/arduino-cli/latest/platform-specification/) describes the `ARDUINO_ARCH_{build.arch}` architecture build parameter. This parameter is missing in the Teensy Arduino cores.

The architecture parameter is normally defined in `platform.txt`. The Teensy cores appear not to use `platform.txt` but that does not mean that the architecture parameter should not be passed to the build process.

The architecture parameter is essential for determining the used architecture in cross-platform applications, so the build process can be customized and account for needs of certain cores. All common Arduino cores define the architecture parameter (except for the Teensy cores).

The lack of definition of the architecture parameter in the Teensy cores prevents consistent checking of the architecture in cross platform applications.

Architecture parameters for the most common Arduino cores are:
ARDUINO_ARCH_AVR
ARDUINO_ARCH_SAMD
ARDUINO_ARCH_MBED
ARDUINO_ARCH_STM32
ARDUINO_ARCH_ESP32
ARDUINO_ARCH_ESP8266

Please add the architecture build parameter to the Teensy cores, and comply to the Arduino Platform Specification.
_(It will take very little efforts to implement.)_

_(The architecture build parameter could be looking something like `ARDUINO_ARCH_NXPKINETIS`, `ARDUINO_ARCH_KINETISL2`, or whatever)_

PaulStoffregen, 2021-05-20:
No plans to do this.

closed 2021-05-20 by PaulStoffregen

lnlp, 2021-05-20:
> No plans to do this.

For what reason?

gpb01, 2021-05-20:
Mmm ... I suppose because it requires some not easy work ... have you seen the name of the folder where the "_cores_" for the teensy are located? Is "hardware/teensy/**avr**" and ... _build.arch_ i based "**_on the architecture folder of the hardware package_**" so, if the _build.arch_ parameter is added in platform.txt, you will receive the **AVR** architecture :-\

This means that the folders structure, within the hardware, folder should be rethought for the various possibilities ...

Just _my 2¢_ 

Guglielmo

PaulStoffregen, 2021-05-20:
> so certain cross-platform applications

Can you be more specific about "certain"?

lnlp, 2021-05-21:
> Can you be more specific about "certain"?

I mean _any_ applications that want to check the architecture build parameter. 
While it will be possible to check for other build flags to detect that code is compiled for Teensy, it's a pitty that the Teensy cores don't provide the standard ARDUINO_ARCH_ build parameter.
Standards and guidelines in general make compatibility, sharing and reuse of code simpler.
For a cross-platform application I was looking to fix a problem. I needed to check certain aspects of the Arduino cores for all supported boards. During the process I observed that the Teensy cores do not generate the ARDUINO_ARCH_ build parameter which made it less easy to make some comparisons.

PaulStoffregen, 2021-05-21:
Again, *which* application?  Be specific.

lnlp, 2021-05-22:
I'm not sure what is not clear about my post above.
For clarity I have added the word "may".

I can't name you a specific application at the moment.
As said, while investigating a problem I ran into I discovered that Teensy cores do not provide the ARDUINO_ARCH_ build parameter. That problem is not related to Teensy, but I made my observation while searching for the problem cause.

FrankBoesing, 2021-05-22:
So, _are_ there any applications that do that?

lnlp, 2021-05-23:
Teensy cores (currently) do not supply the standard ARDUINO_ARCH_xxx build parameter.
I have ran into this and have reported it. I have also motivated why it is better to provide it.
It will be up to you to do something with it or not.
I'm not sure why you keep asking for example applications. I cannot provide you the answer.
I have observed and reported the issue. There probably was reason to invest my time and efforts in reporting it. Whether something is done with it or not is totally up to you.

PaulStoffregen, 2021-05-23:
The answer is still no, exactly the same as when I initially replied and closed this issue.

lnlp, 2021-05-23:
"I have no plans to do that" without any further explanation feels somewhat like 'F... ..o!'.
That any (substantial) explanation is still lacking therefore doesn't surprise me at all.
I don't care you whether you implement it or not. I do care about substantive replies though.

PaulStoffregen, 2021-05-23:
You are not owed any sort of explanation, especially not when you will not name even 1 application.

I'm going to ask you to please understand Robin & I are managing an extraordinarily difficult situation involving labor shortage due to pandemic social distancing requirements while struggling to maintain supplies in the midst of a global semiconductor shortage.  We are far behind on many important tasks.  Continuing to bring this up after a clear answer has been given is not helping.

Please, accept that the answer is no.

lnlp, 2021-05-23:
I understand your situation.
Thanks for the heads up.