PaulStoffregen/cores #562: map() seems kind of broken

mamama1, 2021-05-29:
Hi

on a Teensy 4.0 with latest Teensyduino in PlatformIO I get unexpected behavior with the map function.

`uint16_t retval = map(x, lower, upper, 0, 255);`
shows that the result seems to roll over, as values like 276 are returned for input values of 20470.
It would be expected that whatever "x" contains, the return value of map() would never exceed the range of 0-255.

Thanks for taking a look at it.

dankoman30, 2021-05-29:
Map function does not constrain its output like you might expect. Use the constrain function to clamp your output:

`retval = constrain(retval, 0, 255);`

PaulStoffregen, 2021-05-30:
Indeed, if you check [Arduino's documentation](https://www.arduino.cc/reference/en/language/functions/math/map/) for map(), it says:

> Does not constrain values to within the range, because out-of-range values are sometimes intended and useful. The constrain() function may be used either before or after this function, if limits to the ranges are desired.

closed 2021-09-10 by PaulStoffregen