PaulStoffregen/cores #565: Wire library not working on Teensy 4

deadeert, 2021-06-11:
Hello,

I scratched my head while moving my code from Teensy 3.2 to Teensy 4. 
I used library U8G2 for a SSD1306 screen, on a new hardware designed for the Teensy 4. 

Unit tests ended up to show that Wire library is in fact not working at all. 

I test PIN 18 and 19 with a logical analyzer using following code, and everything goes right : 
```

#include <Wire.h>
void setup(){
  pinMode(18,OUTPUT);
  pinMode(19,OUTPUT); 
  Serial1.begin(9600);
}

void loop(){
  digitalWrite(18,HIGH);
  digitalWrite(19,HIGH);
  delay(1000);
  digitalWirte(18,LOW);
  digitalWrite(19,LOW); 
}
```

I tested teensy4_i2c example " https://github.com/Richard-Gemmell/teensy4_i2c/blob/master/examples/raw/raw_find_slaves/raw_find_slaves.ino", using a logical analyzer and it also worked well. 

So I finished by writing the following MWE example inspired from (https://www.pjrc.com/teensy/td_libs_Wire.html). 

```
#include <Wire.h>

void setup(){
  Serial1.begin(9600);
  Wire.begin(); 
}

void loop(){
  Serial.println("test message");
  delay(1000); 
}
``` 
I don't get any output from the serial line. Removing `Wire.begin()`, I got the right output of the Teensy. 

Investigating the compilation logs, it seems that the proper resolution occurred for Wire library : 
```

/home/d33d34rt/Tools/Arduino/arduino-1.8.15/arduino-builder -dump-prefs -logger=machine -hardware /home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware -hardware /home/d33d34rt/.arduino15/packages -tools /home/d33d34rt/Tools/Arduino/arduino-1.8.15/tools-builder -tools /home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/tools/avr -tools /home/d33d34rt/.arduino15/packages -built-in-libraries /home/d33d34rt/Tools/Arduino/arduino-1.8.15/libraries -libraries /home/d33d34rt/Arduino/libraries -fqbn=teensy:avr:teensy40:usb=serial,speed=528,opt=o2std,keys=en-us -ide-version=10815 -build-path /tmp/arduino_build_569402 -warnings=all -build-cache /tmp/arduino_cache_671384 -verbose /tmp/arduino_modified_sketch_862115/sketch_jun11a.ino
/home/d33d34rt/Tools/Arduino/arduino-1.8.15/arduino-builder -compile -logger=machine -hardware /home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware -hardware /home/d33d34rt/.arduino15/packages -tools /home/d33d34rt/Tools/Arduino/arduino-1.8.15/tools-builder -tools /home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/tools/avr -tools /home/d33d34rt/.arduino15/packages -built-in-libraries /home/d33d34rt/Tools/Arduino/arduino-1.8.15/libraries -libraries /home/d33d34rt/Arduino/libraries -fqbn=teensy:avr:teensy40:usb=serial,speed=528,opt=o2std,keys=en-us -ide-version=10815 -build-path /tmp/arduino_build_569402 -warnings=all -build-cache /tmp/arduino_cache_671384 -verbose /tmp/arduino_modified_sketch_862115/sketch_jun11a.ino
Using board 'teensy40' from platform in folder: /home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr
Using core 'teensy4' from platform in folder: /home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr
Detecting libraries used...
/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=152 -DARDUINO=10815 -DARDUINO_TEENSY40 -DF_CPU=528000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -I/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/cores/teensy4 /tmp/arduino_build_569402/sketch/sketch_jun11a.ino.cpp -o /dev/null
Alternatives for Wire.h: [Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.0]
/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=152 -DARDUINO=10815 -DARDUINO_TEENSY40 -DF_CPU=528000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -I/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/cores/teensy4 -I/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/libraries/Wire /tmp/arduino_build_569402/sketch/sketch_jun11a.ino.cpp -o /dev/null
/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=152 -DARDUINO=10815 -DARDUINO_TEENSY40 -DF_CPU=528000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -I/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/cores/teensy4 -I/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/libraries/Wire -I/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/libraries/Wire/utility /home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/libraries/Wire/Wire.cpp -o /dev/null
/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=152 -DARDUINO=10815 -DARDUINO_TEENSY40 -DF_CPU=528000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -I/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/cores/teensy4 -I/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/libraries/Wire -I/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/libraries/Wire/utility /home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/libraries/Wire/WireIMXRT.cpp -o /dev/null
/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=152 -DARDUINO=10815 -DARDUINO_TEENSY40 -DF_CPU=528000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -I/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/cores/teensy4 -I/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/libraries/Wire -I/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/libraries/Wire/utility /home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/libraries/Wire/WireKinetis.cpp -o /dev/null
/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=152 -DARDUINO=10815 -DARDUINO_TEENSY40 -DF_CPU=528000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -I/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/cores/teensy4 -I/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/libraries/Wire -I/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/libraries/Wire/utility /home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/libraries/Wire/utility/twi.c -o /dev/null
Generating function prototypes...
/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/../tools/arm/bin/arm-none-eabi-g++ -E -CC -x c++ -w -g -Wall -ffunction-sections -fdata-sections -nostdlib -std=gnu++14 -fno-exceptions -fpermissive -fno-rtti -fno-threadsafe-statics -felide-constructors -Wno-error=narrowing -mthumb -mcpu=cortex-m7 -mfloat-abi=hard -mfpu=fpv5-d16 -D__IMXRT1062__ -DTEENSYDUINO=152 -DARDUINO=10815 -DARDUINO_TEENSY40 -DF_CPU=528000000 -DUSB_SERIAL -DLAYOUT_US_ENGLISH -I/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/cores/teensy4 -I/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/libraries/Wire /tmp/arduino_build_569402/sketch/sketch_jun11a.ino.cpp -o /tmp/arduino_build_569402/preproc/ctags_target_for_gcc_minus_e.cpp
[... cut ...]
Utilisation de la bibliothèque Wire version 1.0 dans le dossier: /home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/avr/libraries/Wire 
/home/d33d34rt/Tools/Arduino/arduino-1.8.15/hardware/teensy/../tools/arm/bin/arm-none-eabi-size -A /tmp/arduino_build_569402/sketch_jun11a.ino.elf
Le croquis utilise 13493 octets (0%) de l'espace de stockage de programmes. Le maximum est de 2031616 octets.
Les variables globales utilisent 17021 octets (3%) de mémoire dynamique, ce qui laisse 507267 octets pour les variables locales. Le maximum est de 524288 octets. 
``` 

Have you any idea of what can goes wrong ? 

Thanks

PaulStoffregen, 2021-06-11:
> Have you any idea of what can goes wrong ?

I see a program which uses Serial1.begin() to communicate with the hardware serial port, but then uses Serial.print() which transmits to the USB virtual serial, not the hardware serial port.

Wire definitely does work on Teensy 4.0.  Many thousands of people have used it successfully.

You really need to post a question on the forum to ask for help.  This is not the place.  I'm closing this issue now.

deadeert, 2021-06-11:
Hum about the Serial.print, I used Serial1.print function, I just make a mistake when I was writing this ticket (I didn't copy/paste). 
If you really think that my issue is just I don't know how to use an API you wrong. 
This is not working: 

```
#include <Wire.h>

void setup() {
  Serial1.begin(9600); 
  Wire.begin(); 
  
}

void loop() {
  Serial1.println("test"); 
  delay(1000); 
} 
```
This is working. 

```
#include <Wire.h>

void setup() {
  Serial1.begin()
  
}

void loop() {
  Serial1.println("test"); 
  delay(1000); 
} 
```

Is it better now? 

``` Wire definitely does work on Teensy 4.0. Many thousands of people have used it successfully```

Great, thousands of people are actually experimenting issues on your amazing Teensy board, on various topics. If you think there is nothing special to see, well it is your choice.

Thanks for nothing

PaulStoffregen, 2021-06-11:
Wire absolutely does work.  I personally use it.  Many other people use it too.

PaulStoffregen, 2021-06-11:
Again, if you want help, you need to ask on the forum (not here) and you need to show a complete program which I or anyone else can copy into Arduino and run on a Teensy 4.0 to reproduce the problem.

Please, make sure you copy the exact program.  If you show a wrong program, as above on this issue, it only adds confusion and makes helping you much harder.

I do not know why Wire is not working for you.  I can't see any complete program to reproduce the problem.  But I do know Wire does work, so this issue is closed.  The forum is where we help with projects and how to use libraries.  This issue tracker is only for software defects - and since Wire works when I use it, and it works when so many other people do too, until you can demonstrate a reproducible problem with a complete program on the forum, I'm considering this case closed.

I do still want to help you get Wire working ... just not here.  Tech help / support is done on the forum.

PaulStoffregen, 2021-06-19:
Retesting this today.

It appears to be fixed by recent startup bug fix.
https://github.com/PaulStoffregen/cores/commit/b8478ad56714087692b5c38ba0b0d864b035c2ed

PaulStoffregen, 2021-06-19:
An installer is now available with this bug fix.

https://forum.pjrc.com/threads/67516-Teensyduino-1-54-Beta-10

PaulStoffregen, 2021-06-19:
@deadeert - Any chance you could give 1.54-beta10 a try?  Does the startup bug fix fully resolve this problem?

I tested with the simple code above, and several permutations of it which fail to boot on 1.54-beta9 and 1.53.  All seem to work fine now.

deadeert, 2021-06-23:
Hey, 
Sorry for the delay. 
will give a try as soon as I can. 
thanks for the support.

deadeert, 2021-06-23:
Expect some delays because I’m currently out of the office, but will give a try most probably begin of July.

PaulStoffregen, 2021-06-23:
no worries, whenever is fine

PaulStoffregen, 2021-06-23:
Work is happening right now on better fault handling and crash reporting.  So if you wait a while, we may have 1.54-beta11 with that work included.

deadeert, 2021-06-25:
Great news, give you feedback on my project integration as soon as I can.

PaulStoffregen, 2021-07-09:
Teensyduino 1.54 was released earlier this week.  It has the startup fix.

https://www.pjrc.com/teensy/td_download.html

deadeert, 2021-07-12:
Great, will have a try this week and I let you know.

deadeert, 2021-08-05:
Sorry for the delay, the following MWE is now working (this time I copy / paste) 
``` 
#include <Wire.h>

void setup(){
  Serial1.begin(9600);
  Wire.begin(); 
}

void loop(){
  Serial1.println("test message");
  delay(1000); 
}
``` 

I will make additional testing right now, but it seems that this issue is solved. Well done ! and thanks for the support.

deadeert, 2021-08-05:
I confirm it is not totally working and compatible with my sketches taken from the 3.2. 
Thanks again,  I'm closing the issue now.

closed 2021-08-05 by deadeert