PaulStoffregen/cores #588: std::optional doesn't work

tech-alp, 2021-07-21:
I'm using a Teensy 3.6 microcontroller. I want to use std::optional but I got some errors.
I use c++17.
platform_packages =
  	toolchain-gccarmnoneeabi@~1.90301.0

```txt
src\MainTest\maintest.cpp:19:6: error: 'optional' in namespace 'std' does not name a template type
   19 | std::optional<int> opt{1};
      |      ^~~~~~~~
src\MainTest\maintest.cpp:19:1: note: 'std::optional' is only available from C++17 onwards
Indexing .pio\build\teensy36\lib63e\libHAL.a
   19 | std::optional<int> opt{1};
      | ^~~
src\MainTest\maintest.cpp:20:12: error: 'opt' was not declared in this scope
   20 | auto val = opt.value();
```

FrankBoesing, 2021-07-21:
Teensy uses c++14

tech-alp, 2021-07-21:
Well, When I using std::to_string, I got some errors. Are there relation? to_string come with c++11.

FrankBoesing, 2021-07-21:
https://forum.pjrc.com/threads/66738-to_string()-doesn-t-work