PaulStoffregen/cores #592: Issue with sscanf not parsing correctly

ajraymond, 2021-08-13:
Using:
* Teensy 4.1
* Teensyduino 1.54
* Arduino 1.8.13

I have narrowed down an issue I am having with sscanf to the following snippet of code.

Depending on the order of my variables in the format string and arguments, the results are different.

```
#include <stdio.h>
#include <stdint.h>

void setup() {
    uint8_t i2c_addr = 1;
    uint16_t mem_addr = 1;
    uint8_t len = 1;
    int ret;

    Serial.println(String(i2c_addr) + "," + String(mem_addr) + "," + String(len));
    //prints: 1,1,1

    const char *cmd = "1 2 3\n";
  
    // This is the order I intended to scan the string in
    ret = sscanf(cmd, "%hhu %hu %hhu", &i2c_addr, &mem_addr, &len);
    Serial.println(String(ret) + "- " + String(i2c_addr) + "," + String(mem_addr) + "," + String(len));
    // prints: 3- 1,0,3

    // I just swap the order of the target variables around
    ret = sscanf(cmd, "%hhu %hhu %hu", &i2c_addr, &len, &mem_addr);
    Serial.println(String(ret) + "- " + String(i2c_addr) + "," + String(len) + "," + String(mem_addr));
    // prints: 3- 1,2,3
}

void loop() { }
```

FrankBoesing, 2021-08-13:
Scanf is from the newlib library. If you think there is a bug, report it there.

https://sourceware.org/newlib/

ajraymond, 2021-08-20:
I have started a thread at https://sourceware.org/pipermail/newlib/2021/018546.html

They are asking if newlib is built with C99 support in Teensy.

FrankBoesing, 2021-08-20:
It uses -std=gnu++14

FrankBoesing, 2021-08-20:
Hm, may be it was fixed already... did not think of that, sorry. Teensy uses a pretty old version.

You can try to update to gcc10 (+newer newlib) to see if that changes anything: https://github.com/TeensyUser/doc/wiki/GCC#Switching-between-different-toolchains