PaulStoffregen/cores #607: Current master commit breaks audio library

h4yn0nnym0u5e, 2021-09-11:
Audio/WavFilePlayer example will not compile with **cores 7798d24** and **audio library 1c5426e**

- A complete copy of the code: **Audio/examples/WavFilePlayer/WavFilePlayer.ino**
- Which Teensy board(s) are you using: **Teensy 4.1**
- If other hardware is needed to reproduce the issue: n/a
- Which version of Teensyduino:  **cores 7798d24** and **Arduino 1.8.15** were used, 
- what Tools menu settings **Teensy 4.1 / Serial + MIDI + Audio / 600MHz** were chosen, and 
- which operating system was used: **Windows 10**.  
- If other software is needed, please be specific: n/a

Error messages:
`In file included from E:\Jonathan\Arduino\libraries\Audio/play_sd_raw.h:32:0,
                 from E:\Jonathan\Arduino\libraries\Audio/Audio.h:127,
                 from E:\Jonathan\Arduino\libraries\Audio\examples\WavFilePlayer\WavFilePlayer.ino:28:
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD\src/SD.h:53:7: error: cannot derive from 'final' base 'File' in derived type 'SDFile'
 class SDFile : public File
       ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD\src/SD.h: In member function 'virtual File SDFile::openNextFile(uint8_t)':
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD\src/SD.h:131:41: error: no matching function for call to 'File::File(SDFile*)'
   if (file) return File(new SDFile(file));
                                         ^
In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD\src/SD.h:35:0,
                 from E:\Jonathan\Arduino\libraries\Audio/play_sd_raw.h:32,
                 from E:\Jonathan\Arduino\libraries\Audio/Audio.h:127,
                 from E:\Jonathan\Arduino\libraries\Audio\examples\WavFilePlayer\WavFilePlayer.ino:28:
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:130:2: note: candidate: File::File(const File&&)
  File(const File&& file) {
  ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:130:2: note:   no known conversion for argument 1 from 'SDFile*' to 'const File&&'
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:123:2: note: candidate: File::File(const File&)
  File(const File& file) {
  ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:123:2: note:   no known conversion for argument 1 from 'SDFile*' to 'const File&'
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:113:2: note: candidate: File::File(FileImpl*)
  File(FileImpl *file) {
  ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:113:2: note:   no known conversion for argument 1 from 'SDFile*' to 'FileImpl*'
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:106:12: note: candidate: constexpr File::File()
  constexpr File() : f(nullptr) { }
            ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:106:12: note:   candidate expects 0 arguments, 1 provided
In file included from E:\Jonathan\Arduino\libraries\Audio/play_sd_raw.h:32:0,
                 from E:\Jonathan\Arduino\libraries\Audio/Audio.h:127,
                 from E:\Jonathan\Arduino\libraries\Audio\examples\WavFilePlayer\WavFilePlayer.ino:28:
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD\src/SD.h: In member function 'virtual File SDClass::open(const char*, uint8_t)':
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD\src/SD.h:164:41: error: no matching function for call to 'File::File(SDFile*)'
   if (file) return File(new SDFile(file));
                                         ^
In file included from C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD\src/SD.h:35:0,
                 from E:\Jonathan\Arduino\libraries\Audio/play_sd_raw.h:32,
                 from E:\Jonathan\Arduino\libraries\Audio/Audio.h:127,
                 from E:\Jonathan\Arduino\libraries\Audio\examples\WavFilePlayer\WavFilePlayer.ino:28:
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:130:2: note: candidate: File::File(const File&&)
  File(const File&& file) {
  ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:130:2: note:   no known conversion for argument 1 from 'SDFile*' to 'const File&&'
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:123:2: note: candidate: File::File(const File&)
  File(const File& file) {
  ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:123:2: note:   no known conversion for argument 1 from 'SDFile*' to 'const File&'
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:113:2: note: candidate: File::File(FileImpl*)
  File(FileImpl *file) {
  ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:113:2: note:   no known conversion for argument 1 from 'SDFile*' to 'FileImpl*'
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:106:12: note: candidate: constexpr File::File()
  constexpr File() : f(nullptr) { }
            ^
C:\Program Files (x86)\Arduino\hardware\teensy\avr\cores\teensy4/FS.h:106:12: note:   candidate expects 0 arguments, 1 provided
Multiple libraries were found for "Audio.h"
 Used: E:\Jonathan\Arduino\libraries\Audio
 Not used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Audio
Multiple libraries were found for "SD.h"
 Used: C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
Error compiling for board Teensy 4.1.
`

PaulStoffregen, 2021-09-11:
Please install 1.55-beta2, so you have a consistent set all the recent changes.

https://forum.pjrc.com/threads/68121-Teensyduino-1-55-Beta-2

I'm running WavFilePlayer here on a Teensy 4.1 with all the latest updates (which will soon be 1.55-beta3) and it's playing nicely.

h4yn0nnym0u5e, 2022-05-22:
Fixed, and superseded by later releases anyway

closed 2022-05-22 by h4yn0nnym0u5e