PaulStoffregen/cores #620: map() doesn't work as expected

sstaub, 2021-10-05:
Teensy 4.1, TeensLoader 1.54 on PIO
using following function `int32_t value = map(analogLast, 8, 1008, 0, 100); // convert to 0...100`
gives different out puts on Arduino Raspberry Pi Pico and Teensy 4.1
The maximum value on Arduino is Ok and it is 100, on Teensy the maximum value is 101, which is wrong.

PaulStoffregen, 2021-10-05:
You must show a complete program which I can run on Teensy 4.1 to reproduce the problem.

sstaub, 2021-10-05:
Ok, I see the problem. If the value of analogLast is e.g. 1015 then the output is 101. This is different to the original Arduino map() function, which cut overrides.

PaulStoffregen, 2021-10-05:
Arduino's original map() function does **NOT** constrain the numerical range.

Here is their documentation:
https://www.arduino.cc/reference/en/language/functions/math/map/

You can see the map() function is specifically documented as:

> Does not constrain values to within the range, because out-of-range values are sometimes intended and useful. The constrain() function may be used either before or after this function, if limits to the ranges are desired.

sstaub, 2021-10-05:
I agree, but it is interesting that it differs from the original Arduino function. Looking deeper I see that you use a template.

closed 2021-10-05 by sstaub

PaulStoffregen, 2021-10-05:
Yes, Arduino's map() function has long-standing bugs where it biases the result slightly, and "inverse" map range (both input and output min/max swapped) does not precisely match normal range.  Teensy's version fixes those bugs for integer usage, and offers floating point usage if float input is given.

sstaub, 2021-10-05:
Thanks for the infos. Did you make a PR for the Arduino core api or does your implementation only works for ARMs?

PaulStoffregen, 2021-10-05:
You can see links in the code to the Arduino issues where this has been discussed.  I shared the code with them on this issues, which have been open for many years.

PaulStoffregen, 2021-10-05:
Here's a direct link to the Arduino issue, first reported in 2014.
https://github.com/arduino/ArduinoCore-API/issues/51#issuecomment-69873889