PaulStoffregen/cores #633: Add __irq_enabled() function

mbenkmann, 2022-01-14:
It is better for a function to not simply disable and enable interrupts, so the following pattern is generally better:

    int enabled = __irq_enabled();
    __disable_irq();
    ...
    if (enabled) __enable_irq();

To be able to use this pattern, a function __irq_enabled() is necessary. The following can be inserted right into the relevant headers (kinetis.h for Teensy 3, imxrt.h for Teensy 4) next to __enable_irq() and disable_irq():

    static inline int __irq_enabled() __attribute__((always_inline, unused));
    static inline int __irq_enabled()
    {
        uint32_t pm;
        __asm__ volatile("MRS %0,PRIMASK" : "=r"(pm)::);
        return (pm & 1) == 0;
    };

FrankBoesing, 2022-01-14:
Why not return a bool?

mbenkmann, 2022-01-14:
@FrankBoesing 
I don't know what C standard is supposed to be supported.

FrankBoesing, 2022-01-15:
gnu++14  (C++ 14)

ssilverman, 2023-03-17:
The "atomic" macros can help here, from `<util/atomic.h>`. The syntax looks like:
```c++
ATOMIC_BLOCK(ATOMIC_RESTORESTATE) {
  // Do stuff
}
```