PaulStoffregen/cores #640: Recommend renaming NUM_ENDPOINTS to LAST_ENDPOINT

liudr, 2022-03-26:
Here is the relevant line that uses the #define:

https://github.com/PaulStoffregen/cores/blob/fcece8099baf1081f77d36628f2ac7de5fd70cf4/teensy4/usb.c#L65

The most common problem of NUM_ENDPOINTS is that one has to guess what EPs to include in the count. Should I include EP0 (yes), should I include EP1 (no)? An even more serious problem is if the user decides to skip some EPs and say only use EP3 (IN) and EP4 (OUT). They think they have 2 EPs and based on other definitions they set NUM_ENDPOINTS to 3 to include EP0. But they should have set it to 4 because the skipped EP2 although isn't initialized, MUST be allocated endpoint_t queue heads because how the rest of the code is written: using EP*2 or EP*2+1 to address the queue heads as if all EPs before the EP were present.

PaulStoffregen, 2022-03-30:
While this could be clearer, as a general rule I usually do not change the names of defines or global scope items after the software has been widely published.  Sometimes these changes cause a lot of unexpected pain for people who have modified copies of the code.  To make this sort of change requires a compelling need.

ssilverman, 2022-11-30:
@liudr What about instead adding a comment line to where NUM_ENDPOINTS is defined? This would add some info, while at the same time meeting Paul’s requirements.