PaulStoffregen/cores #652: SerialUSB.read is not functional

ddavis-2015, 2022-09-04:
There appears to be an issue with the USB serial driver supplied with Teensyduino.  Some critical state info. is not being transmitted by the Teensy driver to the Linux host, which in turn causes the Linux cdc_acm driver to block sending any data to the Teensy.

In other words, the SerialUSB only works in one direction: from Teensy to Linux.

The test program is attached.

The test program was also used with an Arduino Nano 33 BLE Sense, and a Raspberry Pi Pico, with no issues.  Same test environment.

Ubuntu Linux 18.04.6
Arduino IDE 1.8.15
Teensyduino 1.57
Board: Teensy4.1
USB Type: Serial
Speed: 600MHz
Optimize: Debug (also Faster tested)
Serial port: /dev/ttyACM0

Steps:
1. Load program onto board
2. Open Arduino IDE Serial Monitor
3. `Echo Test Ready` appears in Serial Monitor
4. In the Serial Monitor input area, type any character and hit `Return`
5. Continue step 4, noticing there is no output in the Serial Monitor.  Eventually the entire Arduino IDE will stop responding until the Teensy is powered off.

This behavior can also been seen at a Terminal command prompt with the Teensy powered on:
```
echo hello > /dev/ttyACM0
```
The command will simply hang until the Teensy is powered off.

[EchoTest.zip](https://github.com/PaulStoffregen/cores/files/9484376/EchoTest.zip)

PaulStoffregen, 2022-09-04:
duplicated on the forum:
https://forum.pjrc.com/threads/71035-Bug-SerialUSB-read-is-not-functional

PaulStoffregen, 2022-09-04:
Works fine when I find it here on Ubuntu 20.04. 
[image]