PaulStoffregen/cores #666: HardwareSerial availableforWrite() using RX head and tail

noisymime, 2022-12-01:
Using a Teensy 3.5 and getting a number of strange lockups when transmitting over hardware serial. The code relies heavily on monitoring the transmit buffer that is available as blocking is not desirable in this case. 

In trying to track down the lockup issue I came across this in the `Serial.availableForWrite()` function:

```
int HardwareSerial::availableForWrite(void)
{
	uint8_t head, tail;

	head = rx_buffer_head;
	tail = rx_buffer_tail;
	if (head >= tail) return TX_BUFFER_SIZE - 1 - head + tail;
	return tail - head - 1;
}
```

Shouldn't `head` and `tail` here be getting set to the `tx` variables rather than the `rx` ones?

PaulStoffregen, 2022-12-17:
availableForWrite() returns the amount of data which can be transmitted without delay.

Documentation here: availableForWrite

PaulStoffregen, 2022-12-17:
Yes, you're right.  I first looked at Teensy 4.0.  Then yes, seems this has been wrong on Teensy 2.0 for a very long time.  Should be fixed now (or next software release).

commit 61b9e203 (2022-12-17): Fix Serial1.availableForWrite() on Teensy 2.0
fixes #666
closed 2022-12-17 by PaulStoffregen