PaulStoffregen/cores #670: Expose ` File` FileImpl

pr8x, 2022-12-21:
Please use this Github issue *only for reporting a specific problem*.

Please **do not use Github issues for project troubleshooting.**  Post on the PJRC Forum for Teensy-related troubleshooting.

https://forum.pjrc.com/forum.php

General suggestions for new features should also be discussed on the forum.  Github issues for new features are usually only appropriate after decisions are made via forum discussion.

Also check out the most common troubleshooting page:

https://www.pjrc.com/teensy/troubleshoot.html

Before reporting any issue, please make sure you are using the latest Teensyduino version.

https://www.pjrc.com/teensy/td_download.html

If a beta test version is available, please check whether it fixes the issue.

When reporting any issue, please try to provide complete information.  The following are almost always needed:

- A complete copy of the code (which can be copied into Arduino and run on a Teensy board) which reproduces the problem is usually needed to investigate any issue.  Please try to trim your program to the minimum necessary, ideally not depending on extra hardware if possible.  But a complete program that can be copied into Arduino without guesswork is essential.  Please do not submit this Github issue with only code fragments or a vague description of code to recreate the issue.

- Which Teensy board(s) are you using.

- If other hardware is needed to reproduce the issue, please provide links or part numbers where this hardware can be purchased.  Very specific details showing how the hardware is connected also help.  Post photos if needed.

- Which version of Teensyduino and Arduino were used, what Tools menu settings were chosen, and which operating system was used.  If other software is needed, please be specific.

PaulStoffregen, 2022-12-21:
No, it's abstraction layer.  It's entire purpose is to not expose this sort of thing.

closed 2022-12-21 by PaulStoffregen

pr8x, 2022-12-21:
How about exposing `isHidden()` on `File`? Not sure how feasible it is to require that interface for all file implementations (might not be supported for some file systems).