PaulStoffregen/cores #677: Teensy 4.1 Startup Sequence

dwotherspoon, 2023-01-12:
Working on a project also using a Teensy 4.1 (baremetal), and was studying the support libraries here. Noticed a strange line in the startup sequence in [teensy4/startup.c:65](https://github.com/PaulStoffregen/cores/blob/master/teensy4/startup.c#L65)

> IOMUXC_GPR_GPR14 = 0x00AA0000;

From the reference manual GPR14 is only used for ACMP, and the value being written doesn't seem to match the field descriptions. The blame for this shows the [commit](https://github.com/PaulStoffregen/cores/commit/24620b1a715b356865c48d075adbb70da96b8c0d) was for dynamic partitioning of TCMs, which shouldn't involve reconfiguring ACMP? Maybe I'm missing something?

FrankBoesing, 2023-01-17:
Not sure which reference manual you're referring to ;-) 

The lower bits are for ACMP, the upper part is for memory configuration.

PaulStoffregen, 2023-01-17:
NXP changed the documentation between Rev 2 and Rev 3.  The code was written based on earlier versions.

[image]

[image]

FrankBoesing, 2023-01-17:
Ok, you should update your homepage, as there is an outdated download link.

dwotherspoon, 2023-01-17:
Thank you, although I wonder why this was removed.

I've been using "i.MX RT1060X Processor Reference Manual, Rev. 1 05/2022" for my project (obtained from NXP's website) and also "i.MX RT1060 Processor Reference Manual, Rev. 3, 07/2021" (obtained from the PJRC website). Going by the dates on the documents, it seems like the IMXRT1060X manual superseded the IMXRT1060 processor reference manual, which is quite confusing.

Neither of these documents appear to mention CM7_CFGITCMSZ or CM7_CFGDTCMSZ...

PaulStoffregen, 2023-01-17:
> although I wonder why this was removed.

me too

dwotherspoon, 2023-01-31:
I asked on [NXP Community](https://community.nxp.com/t5/i-MX-RT/i-MXRT-106X-CM7-CFGITCMSZ-and-CM7-CFGDTCMSZ/m-p/1590068), although not much clearer. I'm understanding that they're either read-only or shouldn't be touched.

PaulStoffregen, 2023-01-31:
haha, yeah, that NXP answer sure cleared everything right up!!

I'm going to close this issue and leave this very widely used and well tested code as it is.

closed 2023-01-31 by PaulStoffregen

PaulStoffregen, 2023-01-31:
I edited the datasheets page to have both rev 2 and rev 3.  Until now it linked to only rev 2.

https://www.pjrc.com/teensy/datasheets.html

The product pages for Teensy 4.0 and 4.1 have links only rev 3 (has been this way for quite some time).

https://www.pjrc.com/store/teensy41.html#tech

https://www.pjrc.com/store/teensy40.html#tech