PaulStoffregen/cores #681: EventResponder: _serialEvent_default not working as expected

tsandmann, 2023-01-21:
I think commit 8b225ef144ef53d57045bfac74401f802337a72e breaks the intended behavior of the `_serialEvent_default` stuff. 

I discovered the following issue when compiling for Teensy 4.1 with gcc 12.2.0 (*): the constant `_serialEvent_default` is set to `0` instead of `1`. (it's the same for the other `_serialEvent`_X_`default` constants.)

The global variable `_serialEvent_default` is weak/public defined in EventResponder.cpp as `extern const uint8_t _serialEvent_default __attribute__((weak)) PROGMEM = 0;` and strong in serialEvent.cpp as `const uint8_t _serialEvent_default PROGMEM = 1;`.
From my understanding `_serialEvent_default` should result in a constant set to `1` to make the whole thing work correctly. Otherwise for example
https://github.com/PaulStoffregen/cores/blob/6d6c215501509ee49cdd222992d1f788ab66a341/teensy4/yield.cpp#L61 is never executed.

I guess the extern (public) definition was introduced as a workaround for [gcc bug 83271](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=83271)? 
But to make the workaround complete (and not changing the behavior of the `yield()` implementation), you have to define `const uint8_t _serialEvent_default PROGMEM = 1;` as well as extern (public) to make the override of the weak definition work. Therefore `extern` must be added to all the `_serialEvent`_X_`default` definitions in `serialEvent`_(USB)X_`.cpp` for teensy3 and teensy4. 

--
(*) It's the same when compiling with gcc 5.4.1 and it should be independent of the gcc version.

ssilverman, 2023-01-25:
When I made that change, I hadn't realized that `extern` was required since they weren't marked as `static`. I assumed that they were `extern` by default. Looking here: https://en.cppreference.com/w/cpp/language/storage_duration, it looks like "no specifier" doesn't mean "static", but `extern` does. Today I learned (more like reinforced).

Yes, the `extern` was added because of that GCC bug you mention.

ssilverman, 2023-01-25:
Does “no namespace” (global namespace, I guess) also count as “namespace scope”, though? I’m not certain it does, since there’s no `namespace` wrapper, according to that link.

ssilverman, 2023-01-25:
Makes sense. Thanks again for the correction.

PaulStoffregen, 2023-04-03:
I removed _serialEvent_default as part of refactoring the yield() optimization.

Closing this issue, as we no longer have _serialEvent_default.

closed 2023-04-03 by PaulStoffregen