PaulStoffregen/cores #719: Request to not include inplace_function (new in 1.59) in Arduino.h, or change its namespace, or allow external SG14_INPLACE_FUNCTION_THROW definition

ssilverman, 2023-09-04:
Summary: My request is to make it a little easier to change the `stdext::inplace_function` behaviour to something that's more expected, even if the default desired Teensudino behaviour is different.

Currently, `inplace_function`, potentially included in a future version of C++, behaves unexpectedly when a function having no definition is called (it does nothing). Expected behaviour for my libraries is similar to `std::function`, where the program would halt. Teensyduino makes this very hard to change, mostly because it forces the current use by including the header in _Arduino.h_ _and_ by disabling the ability to change the behaviour.

Also, by including `inplace_function` in the `stdext` namespace (coupled with the fact it's included in _Arduino.h_), Teensyduino forces different-than-expected behaviour from the actual proposed definition here: https://github.com/WG21-SG14/SG14/blob/master/SG14/inplace_function.h. _And_ it's not possible to change.

While this may be useful for Teensyduino users, it isn't useful for library writers.

In short:
1. There's no way to redefine `SG14_INPLACE_FUNCTION_THROW` like there is in the recommended version here: https://github.com/WG21-SG14/SG14/blob/master/SG14/inplace_function.h
2. The header is included by default in _Arduino.h_
3. Teensyduino uses the namespace `stdext` and changes the `stdext::inplace_function` behaviour.

If any of these three points change, then I could accomplish my requirements: to not use the forced current behaviour of `inplace_function` in my libraries. If I include my own _inplace_function.h_ then the compiler complains that there's a redefinition (because _Arduino.h_ defines it). If I define `SG14_INPLACE_FUNCTION_THROW` myself then Teensyduino's version redefines it. And so on.

I'd like my library to use the `stdext:inplace_function` definition and not the Teensyduino-modified one.

Could Teensyduino make `inplace_function` at least a little more flexible by either:
1. _Not_ including it in _Arduino.h_, _or_
2. Allowing `SG14_INPLACE_FUNCTION_THROW` to be redefined?

See also:
1. https://github.com/PaulStoffregen/cores/pull/717
3. https://forum.pjrc.com/threads/73342-Teensyduino-1-59-Beta-3?p=331087&viewfull=1#post331087

CC: @luni64

PaulStoffregen, 2023-09-04:
The short answer is no.

ssilverman, 2023-09-04:
> The short answer is no.

Ok, but I'd like to at least understand why it's necessary to include _inplace_function.h_ in Arduino.h?

PaulStoffregen, 2023-09-04:
If you really want your library to have callbacks that effectively halt the user's whole program is used before being initialized, you can quite easily static initialize them to a function with an infinite loop or anything else you believe is appropriate.

ssilverman, 2023-09-04:
> If you really want your library to have callbacks that effectively halt the user's whole program is used before being initialized, you can quite easily static initialize them to a function with an infinite loop or anything else you believe is appropriate.

The following program is equivalent, but crashes when the function, `func`, has nothing assigned to it. `stdext::inplace_function` objects, in my view, should duplicate this behaviour or behave similarly. (Maybe crashing is better than halting, like `std::function` does?)

```c++
bool (*func)() = nullptr;

void setup() {
  Serial.begin(115200);
  while (!Serial && millis() < 4000) {
    // Wait for Serial
  }

  Serial.println(CrashReport);

  Serial.println("Starting...");

  func();

  if (func == nullptr) {
    Serial.println("Here");
  }
}

void loop() {
}
```

I still don't understand the reason for treating unassigned functions as "do nothing" functions, and maybe halting is inferior to crashing, but my strong belief is that unassigned functions should not be treated as "do nothing". I agree with @luni64 (https://github.com/PaulStoffregen/cores/pull/717#issuecomment-1705398139) that this will hide potential bugs. Additionally, this is a feature that nobody's really using yet (well, short of the 1.59-beta users), so why restrict it?

In any case, no worries if you don't respond further. I hear that you don't want to make this change, I just think this discussion is important and wanted to get my views into an issue.

ssilverman, 2023-09-04:
Another thought: no matter how trivial a small inconsistency can seem, inconsistencies can lead to unforeseen bugs. And the behaviour is inconsistent, especially with expectations.

PaulStoffregen, 2023-09-04:
The last thing I want to do is keep pouring time into this resolved issue (#717).  If you don't agree with my decision after this (hopefully last) message, I'm asking you to please just let this go.

My 4 basic reasons behind this decision are:

1. Uninitialized "no operation" behavior is meant to allow libraries to create callbacks in a constructor and use them when events occur, without requiring extra checking whether the user actually assigned a function to be called.  It is an intentional feature meant to allow less/simpler code in libraries.
2. Teensy almost always runs real-time applications with little or no user interface.  Halting or abort() is extremely undesirable.
3. We don't have C++ exceptions because we compile with -fno-exceptions.  The std::function behavior is only meaningful in the context of a system with exceptions.
4. We do have a permissive overall philosophy, which is why we're compiling with -fpermissive and -Wno-error=narrowing.

I'm particularly unmoved by the general argument that calling a C function pointer is unsafe.  C++ gives us the means to make well defined behavior for the uninitialized case.  This is an intentional choice to make that defined behavior "safe" and to allow usage in a way which would crash if C function pointers.

However, I do recognize that stdext::inplace_function may someday become part of the C++ STL.  Perhaps we should change the namespace to something indicating Teensy specific?

ssilverman, 2023-09-05:
While I disagree, I accept that the final decision won’t change. I don’t expect you to spend more time on it. I hear the decision. I’ve really tried to make that clear. This is merely a technical discussion now for future readers. Thank you for explaining. (Again, my points here aren’t trying to convince you further. This is for posterity now and future readers.)

I would like to correct a technical inaccuracy, however. `std::function` does not use exceptions if exceptions are turned off. (Per your point no. 3.) On the Teensy, bad functions (eg. nothing assigned to the object) cause a call to the internal `std::__throw_bad_function_call()`, which in turn calls `std::terminate()`. No exceptions are thrown unless exceptions are actually turned on.

I could get behind a namespace change just so the behaviour in the `stdext` namespace isn’t different from the main _inplace_function_ source code. Then I’d agree with your decision.

(To reiterate, I’m not trying to get you to put more time into this or to change your mind; the decision has been made; this really is just a technical discussion for everyone’s betterment.)

PaulStoffregen, 2023-09-05:
Yes, let's change the namespace, just so there's no risk of it ever colliding with future C++ stuff.

Any suggestions on the name?  Would "teensyext" be ok?

luni64, 2023-09-05:
I'd vote for  `teensy` or `pjrc`.  This would be safe and logical and could be used for other/future stuff as well. 
C++17 also allows nested namespaces so you could go for `pjrc::functional` or so,  but I don't think it is necessary to structure the namespace that deep.

luni64, 2023-09-05:
BTW for the EncoderTool I tried to change the namespace from stdext to something else which was quide tedious and required a lot of detail changes in inplace_function.h. I ended up with wrapping my namespace around the whole thing wich was much easier. See here https://github.com/luni64/EncoderTool/blob/master/src/inplace_function.h

ssilverman, 2023-09-05:
> Yes, let's change the namespace, just so there's no risk of it ever colliding with future C++ stuff.
> 
> Any suggestions on the name? Would "teensyext" be ok?

I personally like "teensy" in the name. I like "teensyext" but I also like the "teensy" suggestion.

ssilverman, 2023-09-05:
> BTW for the EncoderTool I tried to change the namespace from stdext to something else which was quite tedious and required a lot of detail changes in inplace_function.h. I ended up with wrapping my namespace around the whole thing wich was much easier. See here https://github.com/luni64/EncoderTool/blob/master/src/inplace_function.h

@luni64 I'm curious, why did it require lots of changes? Isn't it sufficient to just change the `namespace stdext` declaration near the top? What am I missing?

luni64, 2023-09-05:
It is quite some time ago. I remember that it didn't work straight forward when I tried. Might be related to the old compiler. The other additions besides the wrapping namespace where necessary for compliance to c++11 boards

PaulStoffregen, 2023-09-06:
I changed the namespace to "teensy".

ssilverman, 2023-09-06:
I like it.

closed 2023-09-06 by ssilverman